rzeldent/esp32-smartdisplay-demo #1: tone() not found

erich74, 2023-03-31:
Hi,
First off, thanks a lot for your work, I am starting with those Sunton boards and this is really helping.
I have setup a PIO/VSCode project to compile your demo code as a starting point for my project.
I am also there when compiling but I've got this error:

.pio\libdeps\esp32dev\esp32_smartdisplay\src\esp32_smartdisplay.cpp:139:38: error: 'tone' was not declared in this scope

I understand tone() is part of Arduino core library but looks like this is not part of the arduino declination in PIO.

Do you have an idea how to fix it?
Thanks!
Eric

rzeldent, 2023-04-01:
Hi Eric,

Thanks for mentioning it. I'll have to look into this because it worked. Maybe something was taken out of theArduino core.
I'll have to take a look what's going on. 

In the mean time, if you're not using it I suggest you just star out the beep function in esp32_smartdisplay.cpp (in the .pio srouce directory).

Enjoy!

erich74, 2023-04-01:
Thank you very much for your quick answer.
I was planning to use it but I have an 8ohm speaker being shipped so this is not really urgent.
Yes I did comment out this line and compilation was ok.
Thanks again, you are doing an amazing job, I have ordered additionnal boards (different sizes with capacitive touch screen) thanks to your project I can easily setup new project without the complexity and flaws you have sorted out for us!

rzeldent, 2023-04-01:
Still very strange. I re-triggered the build action that is used to rebuild. This worked fine.
Are you using a special library for the esp-32. Have you tried upgrading the packages?

erich74, 2023-04-01:
Here is my platformio.ini, I have not set any additionnal library for the moment:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200

build_flags =
    -Ofast
    #-Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    #-D LV_CONF_PATH=${platformio.include_dir}/lv_conf.h
    -D LV_CONF_PATH=../../../../include/lv_conf.h
    # Smartdisplay settings
    -D TFT_PANEL_ORDER_RGB
    #-D TFT_PANEL_ORDER_BGR
    #-D TFT_ORIENTATION_PORTRAIT
    -D TFT_ORIENTATION_LANDSCAPE
    #-D TFT_ORIENTATION_PORTRAIT_INV
    #-D TFT_ORIENTATION_LANDSCAPE_INV
    -D ESP32_2432S028R
    #-D ESP32_3248S035R
    #-D ESP32_3248S035C

lib_deps = https://github.com/rzeldent/esp32-smartdisplay.git

erich74, 2023-04-01:
and the terminal build output:

Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing ^2.0.0.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 1.0.5+sha.1505cef
|   |-- SPI @ 1.0
|   |-- Wire @ 1.0.1
|   |-- lvgl @ 8.3.4
|   |   |-- ESP32 BLE Arduino @ 1.0.1
Building in release mode
Compiling .pio\build\esp32dev\lib9b9\esp32_smartdisplay\esp32_smartdisplay.cpp.o
.pio\libdeps\esp32dev\esp32_smartdisplay\src\esp32_smartdisplay.cpp: In function 'void smartdisplay_beep(unsigned int, long unsigned int)':
.pio\libdeps\esp32dev\esp32_smartdisplay\src\esp32_smartdisplay.cpp:139:38: error: 'tone' was not declared in this scope
   tone(AUDIO_PIN, frequency, duration);
                                      ^
*** [.pio\build\esp32dev\lib9b9\esp32_smartdisplay\esp32_smartdisplay.cpp.o] Error 1
=========================================================== [FAILED] Took 5.11 seconds ===========================================================

rzeldent, 2023-04-01:
Hi Eric,

I see there are different versions. See below the output from the build action..

```
Platform has been successfully updated
WARNING: This command is deprecated and will be removed in the next releases.
Please use `pio pkg update` instead.
Platform Manager: Updating espressif32 @ 6.0.1
Platform Manager: Removing espressif32 @ 6.0.1
Platform Manager: espressif32@6.0.1 has been removed!
Platform Manager: Installing platformio/espressif32 @ 6.1.0Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%Unpacking
Platform Manager: espressif32@6.1.0 has been installed!
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 is already installed
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.40500.0Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
UnpackingTool Manager: tool-esptoolpy@1.40500.0 has been installed!
Tool Manager: tool-cmake@3.16.4 is already up-to-date
Tool Manager: tool-esptoolpy@1.40500.0 is already up-to-date
Tool Manager: tool-mkfatfs@2.0.1 is already up-to-date
Tool Manager: tool-mklittlefs@1.203.210628 is already up-to-date
Tool Manager: tool-mkspiffs@2.230.0 is already up-to-date
Tool Manager: tool-ninja@1.9.0 is already up-to-date
Tool Manager: tool-openocd-esp32@2.1100.20220706 is already up-to-date
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 is already up-to-date
```

You could try to update your system; try
```pio pkg update```

Maybe this helps...

Kind regards,

Rene

erich74, 2023-04-01:
Hi Rene,

It works now after pio pkg update !
Thanks a lot!
Eric

rzeldent, 2023-04-02:
Great!!

Enjoy!

closed 2023-04-02 by rzeldent