rzeldent/esp32-smartdisplay-demo #4: No display for a 2432S028R

Smanar, 2023-12-13:
Hello, I m trying to make the demo working, but for the moment without success.
The code is running, I can see information on serial monitor, the led is flashing but nothing on display.
I m using the defaut code with some changes on a 2432S028R Hardware.

```
[platformio]
default_envs = esp32dev
#default_envs = esp32-c3-devkitm-1
#default_envs = esp32-s3-devkitc-1-n16-psram

[env]
platform = espressif32
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

# Partition scheme for OTA
board_build.partitions = min_spiffs.csv

build_flags =
    -Ofast
    -Wall
    #-Wno-discarded-qualifiers
    #-Wno-unused-variable
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    -D LV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf.h
    # EspressIf library defines
    -D ESP_LCD_ST7796_VER_MAJOR=1
    -D ESP_LCD_ST7796_VER_MINOR=2
    -D ESP_LCD_ST7796_VER_PATCH=0
    -D ESP_LCD_ILI9341_VER_MAJOR=1
    -D ESP_LCD_ILI9341_VER_MINOR=2
    -D ESP_LCD_ILI9341_VER_PATCH=0
    -D ESP_LCD_GC9A01_VER_MAJOR=1
    -D ESP_LCD_GC9A01_VER_MINOR=2
    -D ESP_LCD_GC9A01_VER_PATCH=0
    -D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1
    -D CONFIG_XPT2046_CONVERT_ADC_TO_COORDS
    -D CONFIG_XPT2046_Z_THRESHOLD=600
    #-D TFT_WIDTH=240
    #-D TFT_HEIGHT=320
    #-D PIN_BCKL=16

#lib_deps = rzeldent/esp32_smartdisplay
lib_deps =
    https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3
    https://github.com/schreibfaul1/ESP32-audioI2S

[env:esp32dev]
board = esp32dev

build_flags =
    ${env.build_flags}
    # Smartdisplay selection
    -D ESP32_2432S028R

lib_deps = ${env.lib_deps}
```

And BTW, I had so much warning during compilation

Mschmid0, 2023-12-13:
Would you consider Adding Support for the  ILI9341_2_DRIVER?

It has an Alternative Init Sequence.
I have a Newer Revision of [ESP32-2432S028R] with USB-C that Requires it. I Got it working with TFTP_eSPI but not with your Lib because the Init Code is way different.

https://github.com/Bodmer/TFT_eSPI/issues/1172

Smanar, 2023-12-14:
So it's definitively a problem with the lib
It work if I use

`https://github.com/rzeldent/esp32-smartdisplay.git@^1.0.6`

rzeldent, 2023-12-14:
Hi Smanar,

Might also be that indeed there's something wrong but now, working on the feature branch 'settings-in-board', just tested the esp32-3248S035; it works. 

Think this might be caused by the demo code; If the wifi does not connect it the audio pointer is null

Use:

```
#ifdef HAS_SPEAKER
    // Connect to WiFi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    if (WiFi.waitForConnectResult() == WL_CONNECTED)
    {
        audio = new Audio(true, I2S_DAC_CHANNEL_LEFT_EN);
        audio->forceMono(true);
        audio->setVolume(10);

        while (!audio->connecttohost(RADIO_URL))
            delay(500);
    }
#endif
```
 and

```
#ifdef HAS_SPEAK
    if (audio)
        audio->loop();
#endif
```

The if (audio) was missing!

rzeldent, 2023-12-14:
Something else I noticed. Maybe due to the extra flash. Problems uploading/crc errors etc, even at low speeds.
Dit a power cycle with BOOT pressed and flashed. I had a black screen after flashing, frozen. But after a power cycle all worked...

Smanar, 2023-12-15:
Same result, have started from scratch.
With the https://github.com/rzeldent/esp32-smartdisplay-demo/tree/feature/esp32s3 version and your changes.

Have just edited

[platformio]
default_envs = esp32dev

and 

```
[env:esp32dev]
board = esp32dev
```

```
build_flags =
    ${env.build_flags}
    # Smartdisplay selection
    -D ESP32_3248S035R
```

Have used the lv_conf.f  from the homepage
And If I try this demo version with old lib version (https://github.com/rzeldent/esp32-smartdisplay.git), I have black screen too

Smanar, 2023-12-16:
So for information I have tried other projects on github.
All project using this libs are broken, the display is working but corrupted (text are like trought a mirror for exemple)
Perhaps a change on the hardware or a conflict with others libs versions ?

I m now trying CYD-Klipper, this one LVLG too but not your lib and all is working.

rzeldent, 2023-12-16:
"text are like trought a mirror" do you mean like this?

[image]

Smanar, 2023-12-16:
Nope, you can read them on your picture, text is "reversed"on mine.
If needed I can make a picture but it's more like this one https://github.com/rzeldent/esp32-smartdisplay/issues/35#issue-1955649807

rzeldent, 2023-12-16:
Hi Smanar,

If it is mirrored you can override the param PANEL_MIRROR_X for the specific device:
On my ESP32_3248S035R this works: PANEL_MIRROR_X=true

Hope this helps.

You can use the feature/esps3 branch for that. Make sure to check out the demo with submodules because settings have been moved to the boards. Also there is still development going on!

Smanar, 2023-12-17:
I can make it working :
- using old lib
- using old demo code
- using `-D TFT_FLIPPEDMIRRORED`
- using `-D TFT_ORIENTATION_LANDSCAPE`

But I think some devices features are not detected, as I have only the "ClickMe" button and the counter, not the picture.

rzeldent, 2023-12-17:
Hi Smanar,

Yes, I think you're using the main branch. That demo has no picture.
I was referring to the feature branch, but this is still under development.

Smanar, 2023-12-17:
>Yes, I think you're using the main branch. That demo has no picture.

Ha ok, so all is working fine with adding thoses settings for my device, I haven't tested other project that use this lib yet as I have already started mine with only basic libs but I think it will be be same problem.

closed 2023-12-17 by Smanar