rzeldent/esp32-smartdisplay-demo #14: cannot compile

Grey-Lancaster, 2024-01-21:
```

Processing esp32-2432S028Rv2 (board: esp32-2432S028Rv2; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-2432S028Rv2.html
PLATFORM: Espressif 32 (6.0.0) > esp32-2432S028Rv2
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.4+sha.5e0cba9
|-- ESP32-audioI2S @ 2.0.0+sha.e6ee006
|-- lvgl @ 8.3.9
Building in release mode
Compiling .pio\build\esp32-2432S028Rv2\libf15\esp32_smartdisplay\esp_lcd_panel_io_3wire_spi.c.o
Compiling .pio\build\esp32-2432S028Rv2\libf15\esp32_smartdisplay\lvgl_st7789.c.o
Compiling .pio\build\esp32-2432S028Rv2\libf15\esp32_smartdisplay\lvgl_st7796.c.o
Compiling .pio\build\esp32-2432S028Rv2\libf15\esp32_smartdisplay\lvgl_xpt2046.c.o
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp_lcd_panel_io_3wire_spi.c:69:88: error: unknown type name 'esp_lcd_panel_io_callbacks_t'
 static esp_err_t panel_io_register_event_callbacks(esp_lcd_panel_io_handle_t io, const esp_lcd_panel_io_callbacks_t *cbs, void 
*user_ctx);
                                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp_lcd_panel_io_3wire_spi.c: In function 'esp_lcd_new_panel_io_3wire_spi':
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp_lcd_panel_io_3wire_spi.c:124:19: error: 'esp_lcd_panel_io_t' {aka 'struct esp_lcd_panel_io_t'} has no member named 'register_event_callbacks'
     panel_io->base.register_event_callbacks = panel_io_register_event_callbacks;
                   ^
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp_lcd_panel_io_3wire_spi.c: At top level:
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp_lcd_panel_io_3wire_spi.c:255:88: error: unknown type name 'esp_lcd_panel_io_callbacks_t'
 static esp_err_t panel_io_register_event_callbacks(esp_lcd_panel_io_handle_t io, const esp_lcd_panel_io_callbacks_t *cbs, void 
*user_ctx)
```

Grey-Lancaster, 2024-01-22:
[image]

AlexSchaeferBs, 2024-01-22:
Same issue here with the board `esp32-8048S050C` 

Posted it in Q&A at https://github.com/rzeldent/esp32-smartdisplay/discussions/106

rzeldent, 2024-01-22:
Use the main

Grey-Lancaster, 2024-01-22:
You mean this?
https://github.com/rzeldent/esp32-smartdisplay

That requires you to know what you are doing.  There is no platformioini file

rzeldent, 2024-01-22:
Take a look at the How to use section.

https://github.com/rzeldent/esp32-smartdisplay-demo

This is a demo project that you can use/adapt

Grey-Lancaster, 2024-01-22:
There is no how to use section at https://github.com/rzeldent/esp32-smartdisplay-demo
There is one at https://github.com/rzeldent/esp32-smartdisplay  but again there is nothing that works with a simple git

I am trying to use https://github.com/rzeldent/esp32-smartdisplay-demo but it gives the error 

.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay@src-ad9971f5ec300a76c1fc174c458dfc60/src/esp_lcd_panel_io_3wire_spi.c:69:88: error: unknown type name 'esp_lcd_panel_io_callbacks_t'

[image]

rzeldent, 2024-01-22:
Yes, that's the "How to use section" I'm refering to.

Try to do a

```
pio update
```

Think your installation is outdated

Grey-Lancaster, 2024-01-24:
Well,  esp32-2432S028Rv2 will compile now, but I get a white screen
esp32-2432S028R    fails with
Building in release mode
Linking .pio\build\esp32-2432S028R\firmware.elf
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028R\lib91d\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x28): undefined reference to `lvgl_lcd_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028R\lib91d\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x2c): undefined reference to `lvgl_touch_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028R\lib91d\libesp32_smartdisplay.a(esp32_smartdisplay.c.o): in function `smartdisplay_init':
F:\Users\greyl\Documents\PlatformIO\Projects\cyd2\esp32-smartdisplay-demo/.pio/libdeps/esp32-2432S028R/esp32_smartdisplay/src/esp32_smartdisplay.c:233: undefined reference to `lvgl_lcd_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\Users\greyl\Documents\PlatformIO\Projects\cyd2\esp32-smartdisplay-demo/.pio/libdeps/esp32-2432S028R/esp32_smartdisplay/src/esp32_smartdisplay.c:240: undefined reference to `lvgl_touch_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-2432S028R\firmware.elf] Error 1

rzeldent, 2024-01-24:
HI Gray,

Can you temporary try the develop branch, a fresh clone and submodules ?
Not stable but easiest to find for me if there are any issues...

rzeldent, 2024-01-24:
> @rzeldent I'm using PlatformIO `6.1.13` which is the latest available release as far as I know. I also switched back to the `main` branch. Still, the issue persists, that while compiling within `esp_lcd_panel_io_3wire_spi.c` there is a unknown type name `esp_lcd_panel_io_callbacks_t`
> 
> <img alt="image" width="998" src="https://private-user-images.githubusercontent.com/60078907/299283356-7837b67e-be48-49d8-a3ee-568c8878a0c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYxMjQ2NjUsIm5iZiI6MTcwNjEyNDM2NSwicGF0aCI6Ii82MDA3ODkwNy8yOTkyODMzNTYtNzgzN2I2N2UtYmU0OC00OWQ4LWEzZWUtNTY4Yzg4NzhhMGM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI0VDE5MjYwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4MmQxNGE2YmVjN2FkZDRiNGU2MWRjMDJiOWI1Yjg4YTFiN2MyNDM1MGFlYTNlN2Q3OTY4ZTc5YTdkNDhlMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.cciv0vdJzgUvWdPghPfO2_lWkEBFsXaV6xxcVkoX_w8">

Hi Alex,

This is probably caused because these interfaces are only available for the ESP32-S3.
This has been fixed in develop and needs to be merged to main.
For now just put a #if0/#endif around this file and maybe also the esp_io_expander

Grey-Lancaster, 2024-01-24:
@rzeldent 
Can you temporary try the develop branch, a fresh clone and submodules ?
Not stable but easiest to find for me if there are any issues...

Sorry Apparently I do not know how

F:\esp>git clone --recursive https://github.com/rzeldent/esp32-smartdisplay/tree/develop
Cloning into 'develop'...
fatal: repository 'https://github.com/rzeldent/esp32-smartdisplay/tree/develop/' not found

rzeldent, 2024-01-24:
git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo.git
git switch develop

Grey-Lancaster, 2024-01-24:
@rzeldent 
Thanks, git clone worked, but..
F:\esp>git switch develop
fatal: not a git repository (or any of the parent directories): .git

Grey-Lancaster, 2024-01-24:
@rzeldent Appreciate your help

F:\esp>cd esp32-smartdisplay-demo

F:\esp\esp32-smartdisplay-demo>git switch develop
Switched to a new branch 'develop'
M       boards
branch 'develop' set up to track 'origin/develop'.

F:\esp\esp32-smartdisplay-demo>

Grey-Lancaster, 2024-01-24:
@rzeldent I don't know.  In the other the 28Rv2 would build but the screen was white
In this both 28R and v2 fail to build
|-- lvgl @ 8.3.9
Building in release mode
Linking .pio\build\esp32-2432S028Rv2\firmware.elf
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028Rv2\lib23e\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x28): undefined reference to `lvgl_lcd_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028Rv2\lib23e\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x2c): undefined reference to `lvgl_touch_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32-2432S028Rv2\lib23e\libesp32_smartdisplay.a(esp32_smartdisplay.c.o): in function `smartdisplay_init': 
F:\esp\esp32-smartdisplay-demo/.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp32_smartdisplay.c:233: undefined reference to `lvgl_lcd_init'
c:/users/greyl/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: F:\esp\esp32-smartdisplay-demo/.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/esp32_smartdisplay.c:240: undefined reference to `lvgl_touch_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-2432S028Rv2\firmware.elf] Error 1

Grey-Lancaster, 2024-01-25:
I installed openHASP (it did not work either) and it was turned wrong but lit up my screen.  Is that your demo behind it from when v2 installed but had a white screen?

[image]

Grey-Lancaster, 2024-01-25:
[image]

Grey-Lancaster, 2024-01-25:
I went back to the project I had that would build Rv2 and i "think" I messed with boards  .json

Is this the best an image will look like in these CYD's ? Same issue I am having in ESPHome, images just do not look as good as they do on a Lilon 2.4 TFT

Thanks,
Grey

[image]

rzeldent, 2024-01-25:
Hi Gray,

Image looks a bit overexposed. Maybe decrease the backlight? Looks not completely okay. Do not have these issues myself...

Yes, touch should work and pressing 'Add 1' will increase the counter (that's behind the nice lady). To test this use a bigger board or remove the lady!

githubnu9, 2024-01-25:
I'm also experiencing the same problem:
```
Archiving .pio\build\esp32-8048S070Cv11\libFrameworkArduino.a
Archiving .pio\build\esp32-8048S070Cv11\lib35e\libESP32-audioI2S.a
Linking .pio\build\esp32-8048S070Cv11\firmware.elf
c:/users/*****/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-8048S070Cv11\lib2d0\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x2c): undefined reference to `lvgl_lcd_init'
c:/users/*****/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-8048S070Cv11\lib2d0\libesp32_smartdisplay.a(esp32_smartdisplay.c.o):(.literal.smartdisplay_init+0x30): undefined reference to `lvgl_touch_init'
c:/users/*****/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\esp32-8048S070Cv11\lib2d0\libesp32_smartdisplay.a(esp32_smartdisplay.c.o): in function `smartdisplay_init':
C:\temp\esp32-smartdisplay-demo/.pio/libdeps/esp32-8048S070Cv11/esp32_smartdisplay/src/esp32_smartdisplay.c:237: undefined reference to `lvgl_lcd_init'
c:/users/*****/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\temp\esp32-smartdisplay-demo/.pio/libdeps/esp32-8048S070Cv11/esp32_smartdisplay/src/esp32_smartdisplay.c:243: undefined reference to `lvgl_touch_init'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-8048S070Cv11\firmware.elf] Error 1
```

I did a clean clone and tried also develop:

rzeldent, 2024-01-25:
git submodule update?

githubnu9, 2024-01-25:
Thanks that worked!

githubnu9, 2024-01-25:
any idea why i get this error when compiling my squreline ui?

src/ui/components/ui_comp_instellingen.c:98:45: note: each undeclared identifier is reported only once for each function it appears in
src/ui/components/ui_comp_menubalk.c: In function 'ui_Menubalk_create':
src/ui/components/ui_comp_menubalk.c:84:46: error: 'lv_font_montserrat_16' undeclared (first use in this function); did you mean 'lv_font_montserrat_14'?

rzeldent, 2024-01-25:
You have to enable it in the lvgl_conf.h.
But a font uses a lot of space!

Grey-Lancaster, 2024-01-25:
@rzeldent FWIW the espHome folks think the reason my image is weak is apparently this board does not have psram and it is using 8bit color  This board has a usb-c connector in the middle as well as micro to the side

[15:31:39][C][psram:020]: PSRAM:
[15:31:39][C][psram:021]: Available: NO

Grey-Lancaster, 2024-01-25:
Does not help me? still gives the toolchain error

F:\espdemo\esp32-smartdisplay-demo>git submodule update

F:\espdemo\esp32-smartdisplay-demo>

githubnu9, 2024-01-25:
touch is no working for me either. im using the 8048S070Cv13. Is there something i need to enable? the Button Add 1 does nothing. also other squareline projects with touch work in sim but do not work on the display

Grey-Lancaster, 2024-01-25:
LOL, I have no clue if touch is working. @rzeldent said remove the lady, but I cannot figure out how
I do not see it in main.cpp and I even renamed lena.png to lena.png1 and it still shows

githubnu9, 2024-01-26:
> LOL, I have no clue if touch is working. @rzeldent said remove the lady, but I cannot figure out how I do not see it in main.cpp and I even renamed lena.png to lena.png1 and it still shows

use squareline to remove her

rzeldent, 2024-01-26:
Made the lady (Lena) 80x80 pixels so should no longer cover the counter.
This is in develop, will merge later.

closed 2024-01-26 by rzeldent