rzeldent/esp32-smartdisplay-demo #15: Touch not working.

githubnu9, 2024-01-26:
Hi,

Touch is not working for me. I'm using the 8048S070Cv13 (but same for my 8048S050R). I checked the touch controller and it is [GT911](https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/assets/datasheets/GT911.pdf). The factory image worked with touch so i know that the touchscreen is not broken.

I tried the example code and pushed the Button ```Add 1```, but it does nothing. I also made another squareline project which gave the same result. Touch does work when testing the project in squareline though.

Did anyone experience the same?

rzeldent, 2024-01-26:
Hi Githubnu9,

I have it now... Before it worked...
Will try to find out why but now working on the develop branch...

githubnu9, 2024-01-26:
hi rzeldent, thank you. I tried develop with these three boards:
```
default_envs = esp32-8048S070C
#default_envs = esp32-8048S070N
#default_envs = esp32-8048S070R
```
but did not work unfortunatly, i will follow this issue for updates

rzeldent, 2024-01-27:
Fixed it, Try the develop

githubnu9, 2024-01-27:
i did try to update the lib and reflash the 3 different 7 inch types. unofrtunatly no touch still. my 5 inch gives a blank screen now btw

rzeldent, 2024-01-27:
Did you also updated the boards submodule?

githubnu9, 2024-01-27:
```
c:\temp\dev2>git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo
Cloning into 'esp32-smartdisplay-demo'...
remote: Enumerating objects: 624, done.
remote: Counting objects: 100% (445/445), done.
remote: Compressing objects: 100% (183/183), done.
remote: Total 624 (delta 254), reused 406 (delta 228), pack-reused 179
Receiving objects: 100% (624/624), 2.39 MiB | 9.00 MiB/s, done.
Resolving deltas: 100% (311/311), done.
Submodule 'boards' (https://github.com/rzeldent/platformio-espressif32-sunton.git) registered for path 'boards'
Cloning into 'C:/temp/dev2/esp32-smartdisplay-demo/boards'...
remote: Enumerating objects: 1051, done.
remote: Counting objects: 100% (874/874), done.
remote: Compressing objects: 100% (318/318), done.
remote: Total 1051 (delta 728), reused 698 (delta 556), pack-reused 177
Receiving objects: 100% (1051/1051), 50.15 MiB | 10.63 MiB/s, done.
Resolving deltas: 100% (778/778), done.
Submodule path 'boards': checked out '3612d153c42fb074d6df0542f4e9b31403c4d570'

c:\temp\dev2>cd esp32-smartdisplay-demo

c:\temp\dev2\esp32-smartdisplay-demo> git switch develop
Switched to a new branch 'develop'
M       boards
branch 'develop' set up to track 'origin/develop'.

c:\temp\dev2\esp32-smartdisplay-demo>git submodule update
Submodule path 'boards': checked out '8918ab4eefe0d78ea890fac27a713f4cc5c993e5'
```

githubnu9, 2024-01-27:
ok it looks like to work with my own squareline project

rzeldent, 2024-01-27:
BTW try the 5". There was a typo in the back porch. Values from another board.

Touch has improved but looks like the touch controllers are not aligned with the pixel resolution. Experienced this myself using a esp32-8048S043C. Touch controller reports:
[   478][I][lvgl_gt911_i2c.c:192] lvgl_touch_init(): GT911 xResolution / yResolution: (480, 272)

for a 800x480!

So the internal configuration of the touch has to be updated to match the resolution. This is for the GT911 drivers...

Hope this will get rid of the problems.

githubnu9, 2024-01-27:
so there seems something odd with the touch registration. i made aproject with a large button to test. if i press very much left of it then it will register as clicked on the button is this something to do with calibration? 
[image]

rzeldent, 2024-01-27:
Might also be swapped... That can be fixed with the TOUCH_SWAP parameters in the json

githubnu9, 2024-01-27:
well that did something lol, now i have to touch on other random places. did not fix it

githubnu9, 2024-01-27:
with the 5 inch I know get 
```
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c: In function 'lvgl_update_callback':
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c:83:61: error: 'TOUCH_SWAP_XY' undeclared (first use in this function); did you mean 'TOUCH_SWAP_X'?
     ESP_ERROR_CHECK(esp_lcd_touch_set_swap_xy(touch_handle, TOUCH_SWAP_XY));
                                                             ^~~~~~~~~~~~~
                                                             ```
I did not enable Touch_SWAP, just a clean download of the develop

rzeldent, 2024-01-27:
update the boards git submodule. That's a new define

githubnu9, 2024-01-28:
I re-downloaded everything:
```
c:\temp\dev2\5>git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo
Cloning into 'esp32-smartdisplay-demo'...
remote: Enumerating objects: 629, done.
remote: Counting objects: 100% (450/450), done.
remote: Compressing objects: 100% (186/186), done.
remote: Total 629 (delta 256), reused 411 (delta 230), pack-reused 179
Receiving objects: 100% (629/629), 2.39 MiB | 1.66 MiB/s, done.
Resolving deltas: 100% (313/313), done.
Submodule 'boards' (https://github.com/rzeldent/platformio-espressif32-sunton.git) registered for path 'boards'
Cloning into 'C:/temp/dev2/5/esp32-smartdisplay-demo/boards'...
remote: Enumerating objects: 1068, done.
remote: Counting objects: 100% (204/204), done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 1068 (delta 193), reused 153 (delta 151), pack-reused 864
Receiving objects: 100% (1068/1068), 50.13 MiB | 11.06 MiB/s, done.
Resolving deltas: 100% (799/799), done.
Submodule path 'boards': checked out '3612d153c42fb074d6df0542f4e9b31403c4d570'

c:\temp\dev2\5>
c:\temp\dev2\5>cd esp32-smartdisplay-demo

c:\temp\dev2\5\esp32-smartdisplay-demo> git switch develop
Switched to a new branch 'develop'
M       boards
branch 'develop' set up to track 'origin/develop'.

c:\temp\dev2\5\esp32-smartdisplay-demo>cd ..

c:\temp\dev2\5>code esp32-smartdisplay-demo
```

7inch:
```
Resolving esp32-8048S070C dependencies...
UnknownBoard: Unknown board ID 'esp32-8048S070C'

```

5 inch:
```
                 from .pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c:1:
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c: In function 'lvgl_update_callback':
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c:83:61: error: 'TOUCH_SWAP_XY' undeclared (first use in this function); did you mean 'TOUCH_SWAP_X'?
     ESP_ERROR_CHECK(esp_lcd_touch_set_swap_xy(touch_handle, TOUCH_SWAP_XY));
                                                             ^~~~~~~~~~~~~
```

rzeldent, 2024-01-28:
Looks like the boards submodule directory is on main, it should also be on develop.

githubnu9, 2024-01-28:
thanks the updating worked with develop. but touch is still not where it is supposed to be. i tried different settings for 
```
      "'-D TOUCH_SWAP_XY=true'",
      "'-D TOUCH_SWAP_X=true'",
      "'-D TOUCH_SWAP_Y=false'",
 ```
but could not make it work

I can flash the 5 inch again but screen stays white

rzeldent, 2024-02-02:
Any progress, what is the status?

githubnu9, 2024-02-06:
7 inch looks good with:
```
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_SWAP_X=false'",
      "'-D TOUCH_SWAP_Y=false'",
```

the 5inch is still blank.. hmmm

rzeldent, 2024-02-07:
Hi,

Regarding the 5 inch, I looked at it again and seems to be according to the schematic.
Only found the Red and Blue swapped in the esp32-8048S050N version but this can not be the issue...

rzeldent, 2024-02-07:
Has been fixed in https://github.com/rzeldent/esp32-smartdisplay/tree/develop

closed 2024-02-07 by rzeldent