rzeldent/esp32-smartdisplay-demo #16: esp32-2432S028Rv2.json No video

Grey-Lancaster, 2024-01-27:
I can build it and the speaker plays music in English, but talks in Dutch :-)

All the pin settings look correct compared to a working espHome build

```
# Setup SPI for the display. The ESP32-2432S028R uses separate SPI buses for display and touch
spi:
  - id: tft
    clk_pin: GPIO14
    mosi_pin: GPIO13
    miso_pin: GPIO12
  - id: touch
    clk_pin: GPIO25
    mosi_pin: GPIO32
    miso_pin: GPIO39

# Setup a pin to control the backlight
output:
  - platform: ledc
    pin: GPIO21
    id: backlight_pwm

# Setup the ili9xxx platform
#
# Display is used as 240x320 by default so we rotate it to 90°
display:
  - platform: ili9xxx
    model: ili9341
    spi_id: tft
    cs_pin: GPIO15
    dc_pin: GPIO2
    id: "my_display"   
    dimensions: 240X320
    color_order: RGB
    transform:
       mirror_y: false
       mirror_x: false
       swap_xy: true
```

Your file snip

```
      "-DARDUINO_ESP32_DEV",
      "-DESP32_2432S028Rv2",
      "-DLCD_WIDTH=240",
      "-DLCD_HEIGHT=320",
      "-DBCKL=21",
      "-DLCD_USES_ILI9341",
      "-DILI9341_SPI_HOST=SPI2_HOST",
      "-DILI9341_SPI_MOSI=13",
      "-DILI9341_SPI_MISO=12",
      "-DILI9341_SPI_SCLK=14",
      "-DILI9341_CS=15",
      "-DILI9341_DC=2",
      "'-D ILI9341_VENDOR_CONFIG=\"(ili9341_vendor_config_t[]){{.init_cmds=(ili9341_lcd_init_cmd_t[]){{.cmd=0xCF,.data=(uint8_t[]){0x00,0xC1,0x30},.data_bytes=3},{.cmd=0xED,.data=(uint8_t[]){0x64,0x03,0x12,0x81},.data_bytes=4},{.cmd=0xE8,.data=(uint8_t[]){0x85,0x00,0x78},.data_bytes=3},{.cmd=0xCB,.data=(uint8_t[]){0x39,0x2C,0x00,0x34,0x02},.data_bytes=5},{.cmd=0xF7,.data=(uint8_t[]){0x20},.data_bytes=1},{.cmd=0xEA,.data=(uint8_t[]){0x00,0x00},.data_bytes=2},{.cmd=0xC0,.data=(uint8_t[]){0x10},.data_bytes=1},{.cmd=0xC1,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0xC5,.data=(uint8_t[]){0x30,0x30},.data_bytes=2,},{.cmd=0xC7,.data=(uint8_t[]){0xB7},.data_bytes=1},{.cmd=0x3A,.data=(uint8_t[]){0x55},.data_bytes=1},{.cmd=0x36,.data=(uint8_t[]){0x08},.data_bytes=1},{.cmd=0xB1,.data=(uint8_t[]){0x00,0x1a},.data_bytes=2},{.cmd=0xB6,.data=(uint8_t[]){0x08,0x82,0x27},.data_bytes=3},{.cmd=0xF2,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0x26,.data=(uint8_t[]){0x01},.data_bytes=1},{.cmd=0xE0,.data=(uint8_t[]){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00},.data_bytes=15},{.cmd=0xE1,.data=(uint8_t[]){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F},.data_bytes=15},{.cmd=0x2B,.data=(uint8_t[]){0x00,0x00,0x01,0x3F},.data_bytes=4},{.cmd=0x2A,.data=(uint8_t[]){0x00,0x00,0x00,0xEF},.data_bytes=4},{.cmd=0x21},{.cmd=0x11,.delay_ms=120},{.cmd=0x29,.delay_ms=1}},.init_cmds_size=23}}\"'",
      "-DLCD_SWAP_XY=false",
      "-DLCD_MIRROR_X=true",
      "-DLCD_MIRROR_Y=false",
```

Grey-Lancaster, 2024-01-27:
```
Compiling .pio\build\esp32-2432S028Rv2\libb38\lvgl\core\lv_obj_draw.c.o
.pio/libdeps/esp32-2432S028Rv2/lvgl/src/core/lv_obj.c: In function 'lv_init':
.pio/libdeps/esp32-2432S028Rv2/lvgl/src/core/lv_obj.c:152:18: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     char * txt = "Á";
```

rzeldent, 2024-01-27:
The last warning is nothing important; that's LVGL.

The settings seem to match indeed.  Have you tried the normal esp32-2432S028?
The board is really a esp32-2432S028?

Grey-Lancaster, 2024-01-27:
Yes I have tried both. I am not sure what I did the other day to make it work.  There are 2 different ..R.json files around. One is 77 lines the other like 114.  I "think" the other day the clone had the small file and I replaced it with the big file and it worked. (I could see Lena).  Now the clone has the big file and if I replace it with the small file, it will not build.

I have a "normal" non usb-c board due in, in a couple of days.  Will see what it does.

[image]

rzeldent, 2024-01-27:
This does not matter. Only relevant if you have a parallel connection without controller.

Grey-Lancaster, 2024-01-28:
@rzeldent 
I have been messing with arduini lvgl code today and got this to work
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main/Examples/LVGL

The only thing interesting possibly is the vendor startup code which is in a diff format but I tried to compare it to yours and nothing stood out to me.

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER       // Generic driver for common displays
#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172

```
#elif defined (ILI9341_2_DRIVER) // Alternative init sequence, see https://github.com/Bodmer/TFT_eSPI/issues/1172

{
writecommand(0xCF);
writedata(0x00);
writedata(0XC1);
writedata(0X30);

writecommand(0xED);
writedata(0x64);
writedata(0x03);
writedata(0X12);
writedata(0X81);

writecommand(0xE8);
writedata(0x85);
writedata(0x00);
writedata(0x78);

writecommand(0xCB);
writedata(0x39);
writedata(0x2C);
writedata(0x00);
writedata(0x34);
writedata(0x02);

writecommand(0xF7);
writedata(0x20);

writecommand(0xEA);
writedata(0x00);
writedata(0x00);

writecommand(ILI9341_PWCTR1); //Power control
writedata(0x10); //VRH[5:0]

writecommand(ILI9341_PWCTR2); //Power control
writedata(0x00); //SAP[2:0];BT[3:0]

writecommand(ILI9341_VMCTR1); //VCM control
writedata(0x30);
writedata(0x30);

writecommand(ILI9341_VMCTR2); //VCM control2
writedata(0xB7); //--

writecommand(ILI9341_PIXFMT);
writedata(0x55);

writecommand(0x36); // Memory Access Control
writedata(0x08); // Rotation 0 (portrait mode)

writecommand(ILI9341_FRMCTR1);
writedata(0x00);
writedata(0x1A);

writecommand(ILI9341_DFUNCTR); // Display Function Control
writedata(0x08);
writedata(0x82);
writedata(0x27);

writecommand(0xF2); // 3Gamma Function Disable
writedata(0x00);

writecommand(0x26); //Gamma curve selected
writedata(0x01);

writecommand(0xE0); //Set Gamma
writedata(0x0F);
writedata(0x2A);
writedata(0x28);
writedata(0x08);
writedata(0x0E);
writedata(0x08);
writedata(0x54);
writedata(0xA9);
writedata(0x43);
writedata(0x0A);
writedata(0x0F);
writedata(0x00);
writedata(0x00);
writedata(0x00);
writedata(0x00);

writecommand(0XE1); //Set Gamma
writedata(0x00);
writedata(0x15);
writedata(0x17);
writedata(0x07);
writedata(0x11);
writedata(0x06);
writedata(0x2B);
writedata(0x56);
writedata(0x3C);
writedata(0x05);
writedata(0x10);
writedata(0x0F);
writedata(0x3F);
writedata(0x3F);
writedata(0x0F);

writecommand(0x2B);
writedata(0x00);
writedata(0x00);
writedata(0x01);
writedata(0x3f);

writecommand(0x2A);
writedata(0x00);
writedata(0x00);
writedata(0x00);
writedata(0xef);

writecommand(ILI9341_SLPOUT); //Exit Sleep

end_tft_write();
delay(120);
begin_tft_write();

writecommand(ILI9341_DISPON); //Display on
}
```

This looks to be what was on it when it arived

[image]

If you need user_setup.h or lv_config.h let me know

Thanks,
Grey

Grey-Lancaster, 2024-01-30:
Did I do it wrong? I get an error building
```
      "-DILI9341_DC=2",
      "'-D ILI9341_DEV_CONFIG_VENDOR_CONFIG=\"(ili9341_vendor_config_t[]){{.init_cmds=(ili9341_lcd_init_cmd_t[]){{.cmd=0xCF,.data=(uint8_t[]){0x00,0xC1,0x30},.data_bytes=3},{.cmd=0xED,.data=(uint8_t[]){0x64,0x03,0x12,0x81},.data_bytes=4},{.cmd=0xE8,.data=(uint8_t[]){0x85,0x00,0x78},.data_bytes=3},{.cmd=0xCB,.data=(uint8_t[]){0x39,0x2C,0x00,0x34,0x02},.data_bytes=5},{.cmd=0xF7,.data=(uint8_t[]){0x20},.data_bytes=1},{.cmd=0xEA,.data=(uint8_t[]){0x00,0x00},.data_bytes=2},{.cmd=0xC0,.data=(uint8_t[]){0x10},.data_bytes=1},{.cmd=0xC1,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0xC5,.data=(uint8_t[]){0x30,0x30},.data_bytes=2,},{.cmd=0xC7,.data=(uint8_t[]){0xB7},.data_bytes=1},{.cmd=0x3A,.data=(uint8_t[]){0x55},.data_bytes=1},{.cmd=0x36,.data=(uint8_t[]){0x08},.data_bytes=1},{.cmd=0xB1,.data=(uint8_t[]){0x00,0x1a},.data_bytes=2},{.cmd=0xB6,.data=(uint8_t[]){0x08,0x82,0x27},.data_bytes=3},{.cmd=0xF2,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0x26,.data=(uint8_t[]){0x01},.data_bytes=1},{.cmd=0xE0,.data=(uint8_t[]){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00},.data_bytes=15},{.cmd=0xE1,.data=(uint8_t[]){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F},.data_bytes=15},{.cmd=0x2B,.data=(uint8_t[]){0x00,0x00,0x01,0x3F},.data_bytes=4},{.cmd=0x2A,.data=(uint8_t[]){0x00,0x00,0x00,0xEF},.data_bytes=4},{.cmd=0x21},{.cmd=0x11,.delay_ms=120},{.cmd=0x29,.delay_ms=1}},.init_cmds_size=23}}\"'",
      "-DLCD_SWAP_XY=false",
```

.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/lvgl_ili9341.c: In function 'lvgl_lcd_init':
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/lvgl_ili9341.c:65:26: error: 'ILI9341_VENDOR_CONFIG' undeclared (first use in this function); did you mean 'ILI9341_PANEL_IO_SPI_CONFIG'?
         .vendor_config = ILI9341_VENDOR_CONFIG};
                          ^~~~~~~~~~~~~~~~~~~~~
                          ILI9341_PANEL_IO_SPI_CONFIG
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/lvgl_ili9341.c:65:26: note: each undeclared identifier is reported only once for each function it appears in
Compiling .pio\build\esp32-2432S028Rv2\lib8d5\WiFi\WiFi.cpp.o

Grey-Lancaster, 2024-01-30:
I got a normal CYD today and the demo works fine with 028R
Lena is still big but she looks better than what I somehow had before on the v2
CPU goes up when I tap add, but I cannot see it.
Radio still talks in Dutch :-)

rzeldent, 2024-02-02:
See: https://github.com/rzeldent/esp32-smartdisplay/discussions/92#discussioncomment-8346895

closed 2024-02-02 by rzeldent

Grey-Lancaster, 2024-02-02:
Thanks, have not tried it yet

[image]

Grey-Lancaster, 2024-02-04:
@rzeldent 
Dev branch said no board but I added one.  Anyway it works.  CPU is higher that the others and the layout is diff.
But, job well done!
Grey

[image]