rzeldent/esp32-smartdisplay-demo #20: No backlight on ESP32-4848S040C_I_Y_1

KevinvOosterhout, 2024-02-10:
hi,

I'm completely new to the ESP32 ecosystem so apology if it's a dumb question.

I've cloned the demo repo as follows
`git clone https://github.com/rzeldent/esp32-smartdisplay-demo.git --recurse-submodules`

After which i went to the platformio.ini and uncommented the following:
`default_envs = esp32-4848S040CIY1`

I've uploaded the code to my device and looked at the serial console. 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
E (126) esp_core_dump_flash: Incorrect size of core dump image: -16776961
[   127][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   391][I][main.cpp:25] setup(): Board: esp32-4848S040CIY1
[   392][I][main.cpp:26] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[   393][I][main.cpp:27] setup(): Free heap: 314444 bytes
[   398][I][main.cpp:28] setup(): Free PSRAM: 8386279 bytes
[   403][I][main.cpp:29] setup(): SDK version: v4.4.6-dirty
[   409][D][esp32_smartdisplay.c:209] smartdisplay_init(): smartdisplay_init
[   416][D][lvgl_st7701_par.c:25] lvgl_lcd_init(): lvgl_lcd_init
[   686][D][esp32_smartdisplay.c:109] smartdisplay_lcd_set_backlight(): smartdisplay_lcd_set_backlight. duty:0.500000
[   686][D][lvgl_gt911_i2c.c:167] lvgl_touch_init(): lvgl_touch_init
[   692][D][lvgl_gt911_i2c.c:179] lvgl_touch_init(): i2c_param_config. host: 0
[   699][D][lvgl_gt911_i2c.c:181] lvgl_touch_init(): i2c_driver_install host: 0
[   705][D][lvgl_gt911_i2c.c:197] lvgl_touch_init(): esp_lcd_new_panel_io_i2c. host: 0
[   714][D][lvgl_gt911_i2c.c:201] lvgl_touch_init(): GT911 productId: 911
[   720][D][lvgl_gt911_i2c.c:202] lvgl_touch_init(): GT911 fwId: 1060
[   726][D][lvgl_gt911_i2c.c:203] lvgl_touch_init(): GT911 xResolution/yResolution: (480, 480)
[   734][D][lvgl_gt911_i2c.c:204] lvgl_touch_init(): GT911 vendorId: 00
[   741][D][lvgl_gt911_i2c.c:248] lvgl_touch_init(): esp_lcd_touch_new_i2c_gt911. host: 0
[   748][D][esp32_smartdisplay.c:36] lvgl_update_callback(): lvgl_update_callback
[ 16599][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16629][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16659][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16689][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16719][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16749][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
[ 16779][D][lvgl_gt911_i2c.c:148] gt911_lvgl_touch_cb(): Pressed at: (260,194), strength: 1070513646
```

So the touchscreen looks to be working as i'm touching it. But the display is completely dark (No backlight), When i shine my flashlight on the screen i can see some faint shapes and text so i suspect the only issue is the backlight not turning on. Did i forget to change something?

GilchristT, 2024-02-12:
@KevinvOosterhout I have exactly the same problem.
I've found that setting the backlight duty to 1 after smartdisplay_init seems to do the needful.

```cpp
void setup()
{
  smartdisplay_init();
  smartdisplay_lcd_set_backlight(1);

  auto disp = lv_disp_get_default();
  ...
```

Hope this helps.

rzeldent, 2024-02-12:
Can you try the develop branch?

KevinvOosterhout, 2024-02-13:
> Can you try the develop branch?
> 
> Maybe similar issue: #21

I've tried the Develop branch but the backlight is not turning on. When adding the smartdisplay_lcd_set_backlight(1); it does work.

rzeldent, 2024-02-22:
Hi Kevin,

You're just running the demo, not using any PWM?

You could try to change the PWN channel or the PWN Freqency. PWM currently at the last available channel. See ```smartdisplay.h```
```
#define PWM_CHANNEL_BCKL (SOC_LEDC_CHANNEL_NUM - 1)
#define PWM_FREQ_BCKL 20000
```

Play a bit with these (not full backlight but 0.5f) and let me know if it helps!

Belinda-Bird, 2024-02-24:
I have exactly the same display, just with 3 relays.

The PWM frequency is too high at 20000Hz I changed it to 600 
`#define PWM_FREQ_BCKL 600`

bordeux, 2024-03-03:
600 mhz also works for me. Except that , the device now giving the sound noise low frequency

winitboonkliang, 2024-03-28:
issue reboot

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42053c4a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   145][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5160][I][main.cpp:29] setup(): Board: esp32-4848S040CIY1
[  5160][I][main.cpp:30] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5161][I][main.cpp:31] setup(): Free heap: 314580 bytes
[  5166][I][main.cpp:32] setup(): Free PSRAM: 8386279 bytes
[  5171][I][main.cpp:33] setup(): SDK version: v4.4.6-dirty
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x4037b6d8
file: ".pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/lvgl_panel_st7701_par.c" line 104
func: lvgl_lcd_init
expression: esp_lcd_panel_swap_xy(panel_handle, DISPLAY_SWAP_XY)

abort() was called at PC 0x4037b6db on core 1

Backtrace: 0x40377eb2:0x3fcebc70 0x4037b6e5:0x3fcebc90 0x40381605:0x3fcebcb0 0x4037b6db:0x3fcebd30 0x42029c3c:0x3fcebd50 0x42029b5e:0x3fcebe50 0x42002898:0x3fcebe70 0x4202d8fe:0x3fcebeb0

ELF file SHA256: 964cf2b16bfe70b0

Rebooting...

rzeldent, 2024-03-28:
LS,

Has to do with the missing functions to swap/mirror on the panels. Was added by mistake and should be working now...
See PR: https://github.com/rzeldent/esp32-smartdisplay/pull/152

closed 2024-03-30 by rzeldent