rzeldent/esp32-smartdisplay-demo #21: Nothing shown on screen, only backlight

paaspaas00, 2024-02-12:
So I've got a ESP32-8048S050C display board. I tried to load this project but all I see is white screen (backlight is working) but nothing shown (no GUI etc). Touchscreen works.

What's the issue? I suspect some library mismatch (dunno, required LVGL or something)

Thank you since now! @rzeldent

rzeldent, 2024-02-12:
Might have something to do with: https://community.blynk.cc/t/esp32-core-version-3-x-x-has-some-breaking-changes/68207/3

  // Setup backlight
  pinMode(GPIO_BCKL, OUTPUT);
  digitalWrite(GPIO_BCKL, LOW);
#if ESP_ARDUINO_VERSION_MAJOR >= 3
  ledcAttach(GPIO_BCKL, PWM_FREQ_BCKL, PWM_BITS_BCKL);
#else
  ledcSetup(PWM_CHANNEL_BCKL, PWM_FREQ_BCKL, PWM_BITS_BCKL);
  ledcAttachPin(GPIO_BCKL, PWM_CHANNEL_BCKL);
#endif

paaspaas00, 2024-02-13:
Ok I can confirm that backlight is working (tried to change the backlight value, it's changing the light accordingly).
But, as stated, no image is shown, all I see is the backlit white screen

paaspaas00, 2024-02-13:
> Ok I can confirm that backlight is working (tried to change the backlight value, it's changing the light accordingly). But, as stated, no image is shown, all I see is the backlit white screen

Maybe that's already solved on library's `develop` branch? Can't test it unfortunately

juracipereira, 2024-02-14:
I have the same problem using the ESP32-8048S043 display

rzeldent, 2024-02-14:
Have you updaed platform io to the latest platform?

https://docs.platformio.org/en/latest/core/userguide/platforms/cmd_update.html

pio pkg platform upate

paaspaas00, 2024-02-15:
@rzeldent 
`pio pkg platform update` 
Is not recognized as a valid command. 
I did 
`pio pkg update`
And It says "already up to date"

rzeldent, 2024-02-18:
Hi Passpaas,

Added some background info: https://github.com/rzeldent/esp32-smartdisplay/issues/130.

paaspaas00, 2024-02-18:
Hi @rzeldent , thanks for looking into the issue.
Dunno if I'm able to check the things you mentioned in the other GitHub issue, but will try.

May the `develop` branch contain stuff that would fix the issue? As I said, can't really try it. If it's stable and feature-complete, would you mind merging it to `master`? (So that I can try it, I've seen you changed some board definitions, maybe it helps).

Meanwhile, this is my board, in case you can have a look if the board PCB layout is somewhat different from yours:
[image]

rzeldent, 2024-02-20:
Hi paaspass,

I merged the develop to master, so you can try that...

However, no significant changes are present for the 8048S050C as I remember.
The layout looks similar to the 8048S070C and as far as I could see the definitions should be correct.

Let me know how it went...

paaspaas00, 2024-02-20:
Hi @rzeldent, no luck unfortunately.
Just found this one https://forum.lvgl.io/t/esp32s3-lvgl-not-displaying-on-screen/13680, is there somehing useful in there? In case, could you have a look? Thank you!

rzeldent, 2024-02-22:
Hi Paaspaas,

I updated the boards definitions file with different parameters for the display. This has been done in the current develop boards repo. See PR: https://github.com/rzeldent/platformio-espressif32-sunton/commit/5cae5f6a8185637484f68885baf7dc6d00aad182

These are available in the develop branch, so after you checked out the project, switch to the develop branch and also make sure you're using the development branch of the board definitions.

I assume you see the backlight? If not, with a flashlight do you see some screen?

Pls let me know if this works.

paaspaas00, 2024-02-23:
Hi @rzeldent , done the steps you mentioned, still white screen, no change from before. White backlight and no image

h3lgi, 2024-02-24:
I have the same issue with ESP32-8048S050C, we discussed it quickly in the `discussions` and the first thoughts were that I bought a weird clone.  Then I borrowed a new display from the link on the board definition page the story is the same, only a white screen is shown. 

I traced all of the PCB routes as far as I could and it looks like everything corresponds to the schematic (of course it is possible that I missed something), but all pins definitions correspond to the pinout on the updated board definitions.

Some notes about both of the boards:
- both of them are V1.2 
- both of them arrived with a working LVGL demo. 

Probably they have a different driver LCD driver but I'm not sure how to check that, so can you guide me a bit what I can do to give you a bit more information about the board itself (unfortunately I have only basic tools) ?

P.S. Since I have two of these boards one of them can be destroyed if it is needed for the investigation.

rzeldent, 2024-02-24:
Can you send the link to the demo program that's working? Just to be sure?
I made some modification to the definitions because found a different definition (LVGL groups) for the pinning.

h3lgi, 2024-02-24:
The demo mentioned above was pre-uploaded by the manufacturer and the display just arrived with it on board. 
So the only thing that I can show is how it looked before I uploaded the smart-display demo code, video is available [HERE](https://youtu.be/fXWmD31bxoM)

h3lgi, 2024-02-24:
I just played with the parameters and: 
by changing the `ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(8*1000000)` to `(16*1000000)` the demo appear to works!! 

I've tested on both branches `develop` and `main` on demo and on boards definitions repos and it worked.
I also tested both boards the one from the boards definition URL and one with unknown Aliexpress source and both just worked. 

Sorry that I'm late with 15 mins and you had to buy this board :/

[image]

h3lgi, 2024-02-24:
One more thing in addition to all the above is that on the `boards` - `develop`  branch it works better, when I'm trying the `boards` - `main` branch sometimes the image shifts horizontally.

rzeldent, 2024-02-24:
Hi h3lgi,

That's very good news. I could not find wrong parameter and not having the board did not help.
I made some changes lately because found a different definition in Squareline with different values for the display; so this helped!

Just to be sure, could you send the configuration that is working so I can update the boards files??

Great job!

h3lgi, 2024-02-24:
You can find the config that worked for me  [HERE](https://gist.github.com/h3lgi/e7caf42644520447091d9e7c7438d207)

rzeldent, 2024-02-24:
Thanks Will put it in the repo!

h3lgi, 2024-02-24:
After changing the `ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ` the lcd did work but with some glitches when I tried to interact with the UI i.e. when it had to render new things.  

I did play a bit more with parameters and put everything to relate to the ST7262 datasheet.
[image]

However, it obviously did not work with 25MHz frequency but I placed half of it i.e. 12.5MHz worked and the glitches are gone. 

So my final setting for `pclk` looks like: 
`-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5 *1000000)` 

And the [GIST](https://gist.github.com/h3lgi/e7caf42644520447091d9e7c7438d207) file is also updated. 

P.S. I also changed the `PORCH` values for vsync and hsync depending to the datasheet.

paaspaas00, 2024-02-24:
@h3lgi just found some demo project here https://www.makerfabs.com/sunton-esp32-s3-5-inch-ips-with-touch.html 
Here is the page with a link to the demo (at the end of page)
https://wiki.makerfabs.com/Sunton_ESP32_S3_5_7_inch_IPS_with_Touch.html
This should be for the 050C board. The config looks like it's setting 12000000 as clk and some other params look similar. Could you have a look @rzeldent and compare it with h3lgi config?

h3lgi, 2024-02-25:
@paaspaas00  unfortunately I do not have time for tests next few days, but can you also test the config that I shared on your display and give some feedback?

paaspaas00, 2024-02-25:
> After changing the `ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ` the lcd did work but with some glitches when I tried to interact with the UI i.e. when it had to render new things.
>
> I did play a bit more [...]

@rzeldent I can confirm it works perfectly on my device, both touchscreen and display! Thanks guys!!

themacboy, 2024-03-02:
Thx guys ! Now me ESP32-8048S043C is working for firts time.

closed 2024-09-08 by rzeldent

luci84tm, 2025-01-07:
Hi there,
I have  running Version	0.7.0-rc13 042fe05 on a new sunton-8048s043c.
What can I do to fix this white screen  ? 

Thanks allot!
Lucian

luci84tm, 2025-01-07:
> 

How did you managed, can you provide some infos ?

luci84tm, 2025-01-07:
> > 
> 
> How did you managed, can you provide some infos ?

> Thx guys ! Now me ESP32-8048S043C is working for firts time.

How did you managed, can you provide some infos ?