rzeldent/esp32-smartdisplay-demo #23: Error in esp32-3248S035C

andersondeoliveiramachado, 2024-02-20:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, A                                                                                        PB: 80000000 Hz
[  5047][I][main.cpp:27] setup(): Board: esp32-3248S035C
[  5047][I][main.cpp:28] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[  5049][I][main.cpp:29] setup(): Free heap: 213376 bytes
[  5054][I][main.cpp:30] setup(): Free PSRAM: 0 bytes
[  5059][I][main.cpp:31] setup(): SDK version: v4.4.5
[  5082][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  5181][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  5185][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  5197][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  5815][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: MACHADO, BSSID: e8:01:8d:e2:4c:b0, Channel: 11, Auth: WPA_WPA2_PSK
[  5817][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  5857][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.1.24
[  5858][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  5861][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.24, MASK: 255.255.255.0, GW: 192.168.1.1
[  5912][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[  6181][I][WiFiClient.cpp:253] connect(): select returned due to timeout 250 ms for fd 48
[  6684][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[  6943][D][esp32_smartdisplay.c:209] smartdisplay_init(): smartdisplay_init
[  6946][D][lvgl_st7796_spi.c:33] lvgl_lcd_init(): lvgl_lcd_init
[  7172][D][esp32_smartdisplay.c:109] smartdisplay_lcd_set_backlight(): smartdisplay_lcd_set_backlight. duty:0.500000
[  7174][D][lvgl_gt911_i2c.c:167] lvgl_touch_init(): lvgl_touch_init
[  7179][D][lvgl_gt911_i2c.c:179] lvgl_touch_init(): i2c_param_config. host: 0
[  7186][D][lvgl_gt911_i2c.c:181] lvgl_touch_init(): i2c_driver_install host: 0
[  7193][D][lvgl_gt911_i2c.c:197] lvgl_touch_init(): esp_lcd_new_panel_io_i2c. host: 0
[  7201][D][lvgl_gt911_i2c.c:201] lvgl_touch_init(): GT911 productId: 911
[  7207][D][lvgl_gt911_i2c.c:202] lvgl_touch_init(): GT911 fwId: 1060
[  7213][D][lvgl_gt911_i2c.c:203] lvgl_touch_init(): GT911 xResolution/yResolution: (320, 480)
[  7221][D][lvgl_gt911_i2c.c:204] lvgl_touch_init(): GT911 vendorId: 00
[  7249][D][lvgl_gt911_i2c.c:248] lvgl_touch_init(): esp_lcd_touch_new_i2c_gt911. host: 0
[  7249][D][esp32_smartdisplay.c:36] lvgl_update_callback(): lvgl_update_callback
[  7272][D][esp32_smartdisplay.c:162] smartdisplay_led_set_rgb(): smartdisplay_led_set_rgb. R:1, G:0, B:0
[  7272][D][esp32-hal-adc.c:194] __analogReadMilliVolts(): eFuse Vref: Supported
[  7278][I][esp32-hal-adc.c:235] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1086

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401bd004  PS      : 0x00060d30  A0      : 0x800f2d04  A1      : 0x3ffd7730
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x00000140  A5      : 0x00000078
A6      : 0x00000000  A7      : 0x00000000  A8      : 0xbef7bef7  A9      : 0x0000bef7
A10     : 0x00000000  A11     : 0x0000013f  A12     : 0x00000015  A13     : 0x00000013
A14     : 0x00000004  A15     : 0x3ffc4d84  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4008a0bc  LEND    : 0x4008a0c7  LCOUNT  : 0x00000000

Backtrace: 0x401bd001:0x3ffd7730 0x400f2d01:0x3ffd7750 0x400f296b:0x3ffd77d0 0x400f4ee1:0x3ffd7800 0x400f6269:0x3ffd78a0 0x401bcc51:0x3ffd7970 0x400d535c:0x3ffd7990 0x400d5ad6:0x3ffd7a40 0x401bc69e:0x3ffd7a70 0x400d31ff:0x3ffd7a90 0x400d33aa:0x3ffd7ab0 0x400da57d:0x3ffd7af0 0x400da649:0x3ffd7b40 0x400da9a2:0x3ffd7bc0 0x400daad3:0x3ffd7bf0 0x400db705:0x3ffd7cc0 0x400e7cd4:0x3ffd7d20 0x400d2a73:0x3ffd7d50 0x401258c5:0x3ffd7da0

ELF file SHA256: d0b02ab034037d0f

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    26][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[
`

rzeldent, 2024-02-20:
Hi

Can you enable in the logging just the logging line with only exception decoder?
This will show the call stack

andersondeoliveiramachado, 2024-02-20:
> Hi
> 
> Can you enable in the logging just the logging line with only exception decoder? This will show the call stack

I'm new to programming with ESP32, how would I do this?

rzeldent, 2024-02-20:
Hi,

Just change the line in platformio.ini from
```
monitor_filters = log2file, time, default, esp32_exception_decoder
```
to
```
monitor_filters = esp32_exception_decoder

```

And try again...

rzeldent, 2024-02-22:
Confirmed.

rzeldent, 2024-02-22:
Hi Anderson,

Thanks for detecting this. I did some investigation and querying the GT911 for its configuration fails, at least on my device.
In the develop branch this should be fixed; if querying for the resolution fails it will use the display resolution.

Could you give it a try in the develop branch?

rzeldent, 2024-02-22:
Did some more retesting. Works but sometimes see it still crashes (and restarts). So it is a bit mitigated.

Think this is caused by the driver that also tries to retrieve the same information (but not what's required for the lVGL).

rzeldent, 2024-09-08:
Maybe related to https://github.com/rzeldent/esp32-smartdisplay-demo/issues/43 ?

closed 2026-08-25 by rzeldent