rzeldent/esp32-smartdisplay-demo #25: Compile error on esp32-8048S043C ...

themacboy, 2024-02-28:
Good morning,

Im not really skilled on this things. Simply git clone following you instructions, then loaded on vcode+plastformio and then run the build option.

Im doing something wrong?

```
 *  Executing task: platformio run --environment esp32-8048S043C 

Processing esp32-8048S043C (board: esp32-8048S043C; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-8048S043C.html
PLATFORM: Espressif 32 (6.5.0) > esp32-8048S043C
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.8+sha.38956ed
|-- ESP32-audioI2S @ 2.0.0+sha.c95aafa
|-- lvgl @ 8.3.11
Building in release mode
Compiling .pio/build/esp32-8048S043C/src/main.cpp.o
Compiling .pio/build/esp32-8048S043C/src/ui/components/ui_comp_hook.c.o
Compiling .pio/build/esp32-8048S043C/src/ui/images/ui_img_lena80x80_png.c.o
Compiling .pio/build/esp32-8048S043C/src/ui/screens/ui_scrMain.c.o
Compiling .pio/build/esp32-8048S043C/src/ui/ui.c.o
src/ui/ui.c: In function 'ui_event_btnCount':
src/ui/ui.c:43:16: warning: unused variable 'target' [-Wunused-variable]
     lv_obj_t * target = lv_event_get_target(e);
                ^~~~~~
Compiling .pio/build/esp32-8048S043C/src/ui/ui_helpers.c.o
Building .pio/build/esp32-8048S043C/bootloader.bin
Generating partitions .pio/build/esp32-8048S043C/partitions.bin
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_disp.c.o
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_event.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_group.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_indev.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_indev_scroll.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_class.c.o
src/main.cpp: In function 'void setup()':
src/main.cpp:49:10: warning: unused variable 'disp' [-Wunused-variable]
     auto disp = lv_disp_get_default();
          ^~~~
src/main.cpp: In function 'void loop()':
src/main.cpp:72:30: warning: format '%d' expects argument of type 'int', but argument 3 has type 'long unsigned int' [-Wformat=]
         sprintf(text_buffer, "%d", now);
                              ^~~~  ~~~
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_draw.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_pos.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_scroll.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_style.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_style_gen.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_obj_tree.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_refr.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/core/lv_theme.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/draw/arm2d/lv_gpu_arm2d.c.o
[... 169 lines cut ...]
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_line.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_objx_templ.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_roller.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_slider.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_switch.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_table.c.o
Compiling .pio/build/esp32-8048S043C/lib98f/lvgl/widgets/lv_textarea.c.o
Compiling .pio/build/esp32-8048S043C/liba41/esp_lcd_touch/esp_lcd_touch.c.o
Compiling .pio/build/esp32-8048S043C/lib1bc/esp_lcd_touch_xpt2046/esp_lcd_touch_xpt2046.c.o
In file included from .pio/libdeps/esp32-8048S043C/esp_lcd_touch/src/esp_lcd_touch.c:17:
.pio/libdeps/esp32-8048S043C/esp_lcd_touch/include/esp_lcd_touch.h:76:14: error: 'CONFIG_ESP_LCD_TOUCH_MAX_POINTS' undeclared here (not in a function); did you mean 'CONFIG_FMB_TCP_PORT_MAX_CONN'?
     } coords[CONFIG_ESP_LCD_TOUCH_MAX_POINTS];
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
              CONFIG_FMB_TCP_PORT_MAX_CONN
*** [.pio/build/esp32-8048S043C/liba41/esp_lcd_touch/esp_lcd_touch.c.o] Error 1
========================================================================================================= [FAILED] Took 65.87 seconds =========================================================================================================

Environment      Status    Duration
---------------  --------  ------------
esp32-8048S043C  FAILED    00:01:05.868
==================================================================================================== 1 failed, 0 succeeded in 00:01:05.868 ====================================================================================================

 *  The terminal process "platformio 'run', '--environment', 'esp32-8048S043C'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

rzeldent, 2024-02-28:
Hi,

What branch are you on? There's some refactoring going on on develop...

themacboy, 2024-02-29:
> Hi,
> 
> What branch are you on? There's some refactoring going on on develop...

Main

themacboy, 2024-02-29:
By the way, my installation of Vcode+platformio is brand new

But after upload black screen and the monitor return:

```
S'està executant la tasca: platformio device monitor --environment esp32-8048S043C 

--- forcing DTR inactive
--- forcing RTS inactive

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/ttyUSB1 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   135][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5144][I][main.cpp:27] setup(): Board: esp32-8048S043C
[  5145][I][main.cpp:28] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5146][I][main.cpp:29] setup(): Free heap: 315044 bytes
[  5151][I][main.cpp:30] setup(): Free PSRAM: 8386279 bytes
[  5156][I][main.cpp:31] setup(): SDK version: v4.4.6-dirty
[  5161][D][esp32_smartdisplay.c:209] smartdisplay_init(): smartdisplay_init
[  5169][D][lvgl_st7262_par.c:23] lvgl_lcd_init(): lvgl_lcd_init
[  5198][D][esp32_smartdisplay.c:109] smartdisplay_lcd_set_backlight(): smartdisplay_lcd_set_backlight. duty:0.500000
[  5198][D][lvgl_gt911_i2c.c:34] lvgl_touch_init(): lvgl_touch_init
[  5204][V][esp_lcd_touch_gt911.c:351] esp_lcd_touch_new_i2c_gt911(): esp_lcd_touch_new_spi_gt911. io:3fceca1c, config:3fcebcf0, handle:3fcebcb0
[  5216][V][esp_lcd_touch_gt911.c:76] gt911_reset(): gt911_reset. tp:3d977b08
[  5229][V][esp_lcd_touch_gt911.c:105] gt911_read_info(): gt911_read_info. tp:3d977b08
[  5231][D][esp_lcd_touch_gt911.c:142] gt911_read_info(): GT911 productId: 911
[  5237][D][esp_lcd_touch_gt911.c:143] gt911_read_info(): GT911 fwId: 1060
[  5244][D][esp_lcd_touch_gt911.c:144] gt911_read_info(): GT911 xResolution/yResolution: (480, 272)
[  5253][D][esp_lcd_touch_gt911.c:145] gt911_read_info(): GT911 vendorId: 00
[  5260][D][esp32_smartdisplay.c:36] lvgl_update_callback(): lvgl_update_callback
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00040836  A0      : 0x0aaaa8aa  A1      : 0x3fcebce0  
A2      : 0x00060a30  A3      : 0x00040023  A4      : 0x3c06a418  A5      : 0x00000024  
A6      : 0x3c06a6b5  A7      : 0x3c075e75  A8      : 0x8203fec8  A9      : 0x0aaaa8aa  
A10     : 0x3d977b08  A11     : 0x00000000  A12     : 0xfffffffe  A13     : 0x00000054  
A14     : 0x3c06a42d  A15     : 0xff000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000044  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffb  

Backtrace: 0x403743bd:0x3fcebce0 0x0aaaa8a7:0x3fcebd00 |<-CORRUPTED

  #0  0x403743bd:0x3fcebce0 in _UserExceptionVector at ??:?
  #1  0x0aaaa8a7:0x3fcebd00 in ?? ??:0
  ```

ELF file SHA256: 697d89a0830df841
```

closed 2024-02-29 by rzeldent

themacboy, 2024-02-29:
hahaha not sure why I used --recursive, and not --recurse-submodules . Sorry!

themacboy, 2024-02-29:
Now compiled perfectly and uploaded, and no error on monitor.

But unfortunatelly on restar screen turn white and nothing more, and that the same problem that move me to this demo. I found the same problem on CYD Klipper with this screen:

https://github.com/suchmememanyskill/CYD-Klipper/issues/54

 Thx for all the help.