rzeldent/esp32-smartdisplay-demo #27: Touch not working on esp32-8048S043C

yaconsult, 2024-03-15:
First of all, thank you for making these wonderful repos available for people who want to learn how to use these cheap boards with PlatformIO!

I am new to all this stuff, so I started with this demo repo.  I have an esp32-8048S043C, so I uncommented that line in platformio.ini and the program builds, uploads and runs, but touch does not work.

I found a closed issue with discussion about a similar problem someone had, and the recommendation was to switch to the development branch, so I tried switching to that and in the lower-left corner of PlatformIO, it does say "develop*" for the branch.  I did a clean, build, and upload, but I get the same result.  The program runs and the screen is being updated, but neither "Add 1", nor "Rotate" buttons do anything.

I know that the touch hardware is good because I made an image of the demo program it ships with and when I flash that image, the touch display works in the LVGL demo it runs.  I can scroll and swipe between screens as expected.

As a newbie, I could very well be doing something wrong or missing some critical point.  Please let me know if I missed something or if you need any additional information.

Thank you!

yaconsult, 2024-03-29:
I deleted everything and tried again today by cloning recursively and switching to development branch as follows:

 1016  rm -fr esp32-smartdisplay-demo/
 1017  git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo
 1018  cd esp32-smartdisplay-demo/
 1019  git switch develop
 1020  git submodule update

I noticed the following errors during the build, but I don't know if they are related to touch not working on my 4.3-inch capacitive display board using: default_envs = esp32-8048S043C in platformio.ini.

src/ui/ui.c: In function 'ui_event_Rotate':
src/ui/ui.c:46:16: warning: unused variable 'target' [-Wunused-variable]
     lv_obj_t * target = lv_event_get_target(e);
                ^~~~~~
src/ui/ui.c: In function 'ui_event_btnCount':
src/ui/ui.c:54:16: warning: unused variable 'target' [-Wunused-variable]
     lv_obj_t * target = lv_event_get_target(e);
                ^~~~~~

/home/me/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c: In function 'uartSetPins':
/home/me/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:153:9: warning: 'return' with no value, in function returning non-void
         return;
         ^~~~~~
/home/me/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:149:6: note: declared here
 bool uartSetPins(uint8_t uart_num, int8_t rxPin, int8_t txPin, int8_t ctsPin, int8_t rtsPin)
      ^~~~~~~~~~~

Thanks!

rzeldent, 2024-03-30:
Hi Yaconsult,

Those are warnings, not errors. These are generated because of the (too) strict warning checks... -Wall
Good to take a look at them but you can safely ignore them...

yaconsult, 2024-03-31:
OK, I see what you mean about them only being warnings, but touch is still not working on this particular CYD 8048S043C.  There are no responses to touches anywhere on the screen when running the demo.  It is not interactive at all.  The refresh rate is being updated, so the display code is fine, but no input.  8-(  Input DOES work with the LVGL demo that the board ships with, so the hardware is good.

Thank you!

rzeldent, 2024-03-31:
Are you using the latest version? With the submodules also at the latest version?
I can not reproduce this; See video below just compiled, develop version...

https://github.com/rzeldent/esp32-smartdisplay-demo/assets/7620231/5960360c-f9fd-44d7-a2d9-6fb423dfbf29

](url)

rzeldent, 2024-06-14:
The touch module is programmed in the factory for the resolution so, should be 800x480. There is code around to modify this but reluctant to put this in based on the definition chosen.
I had the same issue with the 800x270 board that reported values for the 800x480 board. I suspect because only the display is different, the same touch modules are used/mixed during production...

Solution would be to set the resolution correctly...

https://www.reddit.com/r/embedded/comments/15bvkst/incorrect_touchscreen_coordinates_goodix_gt911/?onetap_auto=true&one_tap=true

johndoe8967, 2024-06-14:
I checked it, the resolution isn't programmed at all.
[  5243][E][esp_touch_gt911.c:147] gt911_read_info(): GT911 productId: 911
[  5250][E][esp_touch_gt911.c:148] gt911_read_info(): GT911 fwId: 0x1060
[  5256][E][esp_touch_gt911.c:149] gt911_read_info(): GT911 xResolution/yResolution: (0,0)
[  5264][E][esp_touch_gt911.c:150] gt911_read_info(): GT911 vendorId: 0x00
I also tried to write it to my resolution 800x480, but no effect.

rzeldent, 2024-06-16:
There is a repository to reflash the configuration:

https://github.com/TAMCTec/gt911-arduino/blob/main/TAMC_GT911.cpp

Maybe that can help?

closed 2026-08-25 by rzeldent