rzeldent/esp32-smartdisplay-demo #33: not working on ESP32-2432S022N ?

klkl140, 2024-05-06:
Hello
I tried to get something visible on the ESP32-2432S022N.
This project does not mention this type, but in the lib it is implemented.
Its my first lvgl project, the problem might be on my side.

I get the following error
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (526) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x40083a71:0x3ffe3ab0 0x4008d2a9:0x3ffe3ad0 0x40092911:0x3ffe3af0 0x40115242:0x3ffe3c20 0x40083185:0x3ffe3c50 0x40079306:0x3ffe3c90 |<-CORRUPTED

  #0  0x40083a71:0x3ffe3ab0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008d2a9:0x3ffe3ad0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40092911:0x3ffe3af0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40115242:0x3ffe3c20 in do_core_init at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:320
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:441
  #4  0x40083185:0x3ffe3c50 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:630
  #5  0x40079306:0x3ffe3c90 in ?? ??:0

Any hint to get it working?
Thanks and greetings
Klaus

quozl, 2024-05-25:
51fb8b0 worked fine for me on an ESP32-2432S022C, with PlatformIO 6.1.15.

rzeldent, 2024-06-14:
Strange.
Looks like an upload issue...

E (526) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

Might be that the flash size is smaller on your board. You could try to change the flash size in the board definition in the boards directory...

rzeldent, 2024-09-08:
Did this work? Maybe some esp32's with different flash sizes were used...

closed 2026-08-25 by rzeldent