rzeldent/esp32-smartdisplay-demo #34: How to open SquareLine_Project.spj?

openshwprojects, 2024-05-25:
I've downloaded SquareLine Studio 1.4.0, free trial version. I can open example projects from it, but when I try to open SquareLine_Project.spj via "Import Project", it does nothing. No error, no crash, no reaction.
What's wrong?
[image]

uncovery, 2024-05-30:
Same here, would love to get that to work.

Fabio-Zigliani, 2024-07-25:
I have created a new SquareLine project and saved in the same folder realising that there is a second file (.sll) that is missing fomr the repo for us ot be able to open the project. @rzeldent  could you add the original file?

rzeldent, 2024-09-08:
I could not upload it but here is the contents:

```
{
    "name": "SquareLine_Project.spj",
    "depth": 1,
    "width": 320,
    "height": 240,
    "rotation": 0,
    "offset_x": 0,
    "offset_y": 0,
    "shape": "RECTANGLE",
    "multilang": "DISABLE",
    "description": "",
    "board": "CMake/Eclipse/VScode with SDL for development on PC",
    "board_version": "v2.0.0",
    "editor_version": "1.4.2",
    "image": "/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAgGBgcGBQgHBwcJCQgKDBQNDAsLDBkSEw8UHRofHh0aHBwgJC4nICIsIxwcKDcpLDAxNDQ0Hyc5PTgyPC4zNDL/2wBDAQkJCQwLDBgNDRgyIRwhMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjIyMjL/wAARCADwAUADASIAAhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwD3aiiigAopm9mYrGhcjrzgUuJ/+eH/AI+KAHUU3E//ADw/8fFGJ/8Anh/4+KAHUU3E/wDzw/8AHxRif/nh/wCPigB1FNxP/wA8P/HxRif/AJ4f+PigB1FNxP8A88P/AB8UYn/54f8Aj4oAdRTcT/8APD/x8UYn/wCeH/j4oAdRTcT/APPD/wAfFGJ/+eH/AI+KAHUU3E//ADw/8fFGJ/8Anh/4+KAHUU3E/wDzw/8AHxRif/nh/wCPigB1FNxP/wA8P/HxRif/AJ4f+PigB1FNxP8A88P/AB8UYn/54f8Aj4oAdRTcT/8APD/x8UYn/wCeH/j4oAdRTcT/APPD/wAfFGJ/+eH/AI+KAHUU3E//ADw/8fFGJ/8Anh/4+KAHUU3E/wDzw/8AHxRif/nh/wCPigB1FNxP/wA8P/HxRif/AJ4f+PigB1FNxP8A88P/AB8UYn/54f8Aj4oAdRTcT/8APD/x8Um9lYLIhQnpzkUAPooooAKKKKAHWY/0ZT3OSfzqeoLP/j1T8f51JLIsMTyucIilmPoBQA+ivLtT8fXst232JVSEHC7ieR+BFU/+E51j+/H+b/8AxVejHKsTJXt+J50s0w0Xa/4HrtFeRf8ACc6x/fj/ADf/AOKo/wCE51j+/H+b/wDxVV/ZOJ7L7xf2thu7+49doryL/hOdY/vx/m//AMVR/wAJzrH9+P8AN/8A4qj+ycT2X3h/a2G7v7j12ivIv+E51j+/H+b/APxVH/Cc6x/fj/N//iqP7JxPZfeH9rYbu/uPXaK8i/4TnWP78f5v/wDFUf8ACc6x/fj/ADf/AOKo/snE9l94f2thu7+49doryL/hOdY/vx/m/wD8VT4fHerJKrPsZR1ALc/mT/Kk8pxPZfeCzXDd39x61RWdomqxazpcd5FxuyrD0I6itGvOlFxbi90ejGSklJbMKKKKQwor5g/4aO8Yf9A3Q/8AvxN/8dpR+0b4v/6Buh/9+Jf/AI7TUW9gPp6ivmIftGeL/wDoHaH/AN+Jf/jtL/w0X4v/AOgdof8A34l/+O1qqE3sK6PpyivmQftFeLv+gdof/fiX/wCO0o/aJ8Xf9A7RP+/Ev/x2tFg6r2FzI+mqK+Zx+0R4u/6B2if9+Jf/AI7Sj9ofxb/0DtE/78S//HK1WW13tb7w50fS9FfNI/aG8W/9A7RP+/Ev/wAcpR+0L4s/6B2i/wDfmX/45WqyjEvZL7xe0ifStFfNg/aE8Wf9A/Rf+/Mv/wAcpR+0F4s/6B+i/wDfmX/45WiyLFvZL7xe1ifSVFfN3/DQXiv/AKB+i/8AfmX/AOOUo/aB8V/9A/Rf+/Mv/wAcrVcO457JfeL20T6Qor5wH7QHiv8A6B+jf9+Zf/jlKPj/AOKv+gfo3/fmX/45Wi4YzB7JfeHtoH0dUF4P9GY9xgj86qeHtQl1bw1pWpTqizXdnDPIsYIUM6BiBkk4yfWrd5/x6v8Ah/OvAnBwk4vdGq1G0UUVIBRRRQA6z/49U/H+dVte48O6n/16S/8AoBqzZ/8AHqn4/wA6ra9/yLup/wDXpL/6Aa0pfxI+qIq/BL0Z4bRRRX3B8OFFFPaKRIklaNxG5IRypw2OuD3xmgYyirCWF5JcC3S0naYqGEYjJYgjIOOuMU0Wly119lFvKbjO3ygh359MdaXNHuPll2IaKspp19IZQllcMYTiULEx2f73HH40i2N47zItrOzQZ80CMkx467vT8aXPHuHJLsV6KsJY3klq1ylrO1uv3pVjJQfU9Kr1SaewmmtwooooEeo/Df8A5F24/wCvtv8A0BK7GuO+G3/Iu3H/AF9t/wCgJXY18dj/APeZ+p9hgf8AdoegUUUVyHWfAApwpAKcFY/wn8q0g0JiinUCNz/A35U8Qyn/AJZv/wB8mu2nOPchtDRThThbzH/ljJ/3yaeLa4P/ACwl/wC+DXbTqwX2kQ5LuMFOFSC0uT/y7y/98Gniyuj/AMu03/fs12069L+ZfeQ5x7kQpwqYWF4f+XSf/v2aeNPvT/y53H/fpv8ACu2niaK+2vvRDqQ7kApwqwNNvz/y5XP/AH6b/CnjS9QP/Ljdf9+W/wAK7aeLw6/5eL70Q6kO6Kopwq0NJ1I/8w+6/wC/Lf4U8aRqZ/5h13/34b/Cu2njcMv+XkfvRDq0/wCZfeVBThVwaNqh/wCYbef9+G/wp40TVj/zC73/AMB3/wAK7qePwn/P2P8A4Ev8yHWp/wAy+8+uPBf/ACInh7/sGW3/AKKWtW8/49X/AA/nWZ4OjeLwRoEciMjrpturKwwQRGuQRWnef8er/h/OvyXEtOtNru/zPVjshtFFFYjCiiigB1n/AMeqfj/Oq2vf8i7qf/XpL/6Aas2f/Hqn4/zqtr3/ACLup/8AXpL/AOgGtKX8SPqiKvwS9GeG0UUV9wfDhXTxXP2bw9pDDTYL1vOmIEqu23BXoFIHPvnpXMVbt9U1C1h8m3v7qGLrsjmZR+QNZVqbmlbp/k0bUaig3fr/AJo6W/A8/VLOO7FpeXLQ3DC4l2kgqS0Rc4AwzDrjp7U+1H2q7dI7uCa4ttKkhluPOCgud2MEnkKCAW9vSuNZizFmJLE5JJ5Jp0cskRYxyOhZSrbWIyp6g+1Y/VXy2vr/AMCxssUua7X9XudFpTK9uLa5mgBtrtpBIL5YmDHALZwQ4+Xgjnr61Yvw17omqXttPEsU2pNKMzKjMm1+ME5yey9TXJU/zZPJMPmP5Rbds3Hbu6Zx61Tw3vcyZMcR7vK0dc1xpt1HbXD3csNtHarGVivlRkwmGURbCTk55zg5zkVx1FFaUqKp3szOrVdS10FFFFamJ6j8Nv8AkXbj/r7b/wBASuxrjvht/wAi7cf9fbf+gJXY18dj/wDeZ+p9hgf92h6BRRRXIdZ8Bp1qzHVZOtWY6zmZTLMfarUdVY+1Wo65pnJMsx9qtR1Vj7VajrlmcdQsx1ajqrHVqOuaZxTLMfarMdVo+1WY65pnJMtR1ajqrHVqOuWZxVCzHVqOqsdWo65ZnHMsx9qtR9aqx9qtR9a5pHHUPYdH/wCQJYf9e0f/AKCKmvP+PV/w/nUOj/8AIEsP+vaP/wBBFTXn/Hq/4fzr9Bofw4+iP1PDfwYei/IbRRRWpsFFFFADrP8A49U/H+dVtdBPh7UwOT9kl/8AQDVmz/49U/H+dTMqupVgCpGCD0IqoS5ZKXYmceaLXc+fqK73UvhvO10z6bdQiBjkJOWBT2yAc1T/AOFbax/z82P/AH8f/wCJr66OYYaSvznycsvxMXblOOorsf8AhW2sf8/Nj/38f/4mj/hW2sf8/Nj/AN/H/wDiaf1/Dfzon6jif5GcdRXY/wDCttY/5+bH/v4//wATR/wrbWP+fmx/7+P/APE0fX8N/Og+o4n+RnHUV2P/AArbWP8An5sf+/j/APxNH/CttY/5+bH/AL+P/wDE0fX8N/Og+o4n+RnHUV2P/CttY/5+bH/v4/8A8TR/wrbWP+fmx/7+P/8AE0fX8N/Og+o4n+RnHUV2P/CttY/5+bH/AL+P/wDE0+H4a6mZVE15aJH3ZCzEfgQP50PH4ZfbQ1gMS/sM3vhuCPDs/vdt/wCgJXYVT0vTLfSNOisrYHy4x1PViepNXK+VxNVVa0prZs+ow1N0qMYPdIKKKKwNz4BU4NTo5HpUA61ItTIuMU9yyszDsKnS5cdlqotSrWMoo6YYalLeJcS7kHZfyqwl9KP4U/I1QWpVrGUI9jqhgMNLeCNBNQmH8Kfkf8anTU5h/DH+R/xrNWpVrKVOPY6YZTgnvTRppqs4/gj/ACP+NTprFwP4IvyP+NZS1KtYypQ7HVDI8ulvRRrLrdyP4IvyP+NTprt0P+WcP5H/ABrGWpl/pWMqNPsdUOHMqlvQibS+ILsf8s4f++T/AI1MniK7H/LOD/vk/wCNYa1KtYyoU+x0w4VyaW+Hiby+Jbwf8soP++T/AI1Onie9H/LK3/75b/GufWpVrGWHpfynVDg/Ipb4WP8AXzPqLw5K0/hfSZmADSWULEDpkoDVy8/49X/D+dUPC3/IoaL/ANeEH/otav3n/Hq/4fzr6imrQR+fVoRp1ZQirJNpDaKKKsyCiiigB1n/AMeqfj/Op6gs/wDj1T8f51PQAUUUUAFFFFABRRRQAUUUUAFFFFABRRRQAUUUUAFFFFAHxCPAni//AKFTXP8AwXTf/E1IvgXxf/0Kuuf+C+X/AOJr7aopNFRlY+KV8DeLv+hW1v8A8F8v [... cut ...]
    "projectExportFolderPath": "C:\\Users\\rzeld\\source\\repos\\esp32-smartdisplay-demo",
    "backup_cnt": 79,
    "autosave_cnt": 0,
    "lvgl_version": "9.1.0",
    "callfuncsexport": "CPP_FILE",
    "imageexport": "SOURCE",
    "lvgl_include_path": "lvgl.h",
    "drive_stdio": "-",
    "drive_stdio_path": "",
    "drive_posix": "-",
    "drive_posix_path": "",
    "drive_win32": "-",
    "drive_win32_path": "",
    "drive_fatfs": "-",
    "drive_fatfs_path": "",
    "naming": "",
    "naming_force_lowercase": false
}
```

If this works I can add it to the repo...

Fabio-Zigliani, 2024-09-09:
Hi, I can confirm that by adding the second (.sll) file to the folder the Squareline Project Opens fine in Squareline Studio 1.4.2 :)
Fabio

BenjaminPelletier, 2024-12-04:
I can confirm that SquareLine_Studio 1.5.0 does nothing (fails silently; boo) attempting to import SquareLine_Project.spj as is currently in the repo.  However, adding SquareLine_Project.sll populated with the content in the comment above makes it possible to import SquareLine_Project.spj and then open that project successfully.

rzeldent, 2024-12-05:
Hi Benjamin,.

I moved to EEZ studio. Take a look at the development branch!

closed 2026-08-25 by rzeldent