rzeldent/esp32-smartdisplay-demo #35: How to use this demo ?

eagl1, 2024-06-15:
Hi,

I don't know where to run the git command you provided:

`git clone --recurse-submodules`

I opened the project in platformio, but don't know where is the location to run the command.

I build the code and got this result:

```

 *  Executing task: C:\Users\eagle\.platformio\penv\Scripts\platformio.exe run --environment esp32-8048S043C 

Processing esp32-8048S043C (board: esp32-8048S043C; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------UnknownBoard: Unknown board ID 'esp32-8048S043C'

 *  The terminal process "C:\Users\eagle\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32-8048S043C'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

rzeldent, 2024-06-16:
Hi Abdul,

You have to run the command ```git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo.git``` in a normal command window from the prompt.

Make sure you have installed git.

Good luck

rzeldent, 2024-06-16:
Hi Abdul,

You need to do this before opening the project. Remove everything from the directory, run the command above and then op it in visual studion code.

Good luck!

eagl1, 2024-06-16:
I did it and I get this result:

Microsoft Windows [Version 10.0.19045.4529]
> (c) Microsoft Corporation. All rights reserved.
>
> E:\projects_electronics_programming\tft\platformio projects\New folder>git
> clone --recurse-submodules
> fatal: You must specify a repository to clone.
>
> usage: git clone [<options>] [--] <repo> [<dir>]
>
>     -v, --[no-]verbose    be more verbose
>     -q, --[no-]quiet      be more quiet
>     --[no-]progress       force progress reporting
>     --[no-]reject-shallow don't clone shallow repository
>     -n, --no-checkout     don't create a checkout
>     --checkout            opposite of --no-checkout
>     --[no-]bare           create a bare repository
>     --[no-]mirror         create a mirror repository (implies --bare)
>     -l, --[no-]local      to clone from a local repository
>     --no-hardlinks        don't use local hardlinks, always copy
>     --hardlinks           opposite of --no-hardlinks
>     -s, --[no-]shared     setup as shared repository
>     --[no-]recurse-submodules[=<pathspec>]
>                           initialize submodules in the clone
>     --[no-]recursive ...  alias of --recurse-submodules
>     -j, --[no-]jobs <n>   number of submodules cloned in parallel
>     --[no-]template <template-directory>
>                           directory from which templates will be used
>     --[no-]reference <repo>
>                           reference repository
>     --[no-]reference-if-able <repo>
>                           reference repository
>     --[no-]dissociate     use --reference only while cloning
>     -o, --[no-]origin <name>
>                           use <name> instead of 'origin' to track upstream
>     -b, --[no-]branch <branch>
>                           checkout <branch> instead of the remote's HEAD
>     -u, --[no-]upload-pack <path>
>                           path to git-upload-pack on the remote
>     --[no-]depth <depth>  create a shallow clone of that depth
>     --[no-]shallow-since <time>
>                           create a shallow clone since a specific time
>     --[no-]shallow-exclude <revision>
>                           deepen history of shallow clone, excluding rev
>     --[no-]single-branch  clone only one branch, HEAD or --branch
>     --no-tags             don't clone any tags, and make later fetches not
> to follow them
>     --tags                opposite of --no-tags
>     --[no-]shallow-submodules
>                           any cloned submodules will be shallow
>     --[no-]separate-git-dir <gitdir>
>                           separate git dir from working tree
>     --[no-]ref-format <format>
>                           specify the reference format to use
>     -c, --[no-]config <key=value>
>                           set config inside the new repository
>     --[no-]server-option <server-specific>
>                           option to transmit
>     -4, --ipv4            use IPv4 addresses only
>     -6, --ipv6            use IPv6 addresses only
>     --[no-]filter <args>  object filtering
>     --[no-]also-filter-submodules
>                           apply partial clone filters to submodules
>     --[no-]remote-submodules
>                           any cloned submodules will use their
> remote-tracking branch
>     --[no-]sparse         initialize sparse-checkout file to include only
> files at root
>     --[no-]bundle-uri <uri>
>                           a URI for downloading bundles before fetching
> from origin remote
>
>
> E:\projects_electronics_programming\tft\platformio projects\New folder>
>

eagl1, 2024-06-16:
Another try in the demo folder:

[image: image.png]

rzeldent, 2024-06-16:
If you carefully read the output you see:
E:\projects_electronics_programming\tft\platformio projects\New
> folder>git clone --recurse-submodules
> fatal: **You must specify a repository to clone.**

so you did not specify the url. Type (see above):

git clone --recurse-submodules **https://github.com/rzeldent/esp32-smartdisplay-demo.git**

eagl1, 2024-06-16:
Yep, it worked :)

[image: image.png]

So I'm new to projects done in platformio, certainly it's more advanced
than Arduino IDE.

I want to do two tasks:
1. Do my own GUI project, on the platformio IDE
2. Copy the necessary source code and embed it in the `TFT_eSPI` library
because this display isn't supported there. And it will be a difficult task
to do because the coding system in this project and in the TFT_eSPI is
totally different. I hope you can help me with that. And I will pay you.

Regards,

closed 2024-06-30 by eagl1