rzeldent/esp32-smartdisplay-demo #37: Adding new 7" RGB display ESP32 EA7013

eagl1, 2024-06-30:
Hi,

I received this ESP32-EA7013 display from Aliexpreess, I tried to match its settings with the similar RGB display `ESP32-8048S043C`, as I contacted the manufacturing source:

[https://www.eya-display.com/ea7013-s3/](url)

And I asked them for resource files and they gave me the schematic, general manual for the display and ESP-IDF code.

They even didn't tell me what driver IC the display is using and they told me becuase it's RGB display it doesn't need configuration initialization commnads, but from the schematic it's similar to the one used in `ESP32-8048S043C`.

Now in your platformio, how to copy another board package and edited it to fit the display ?

rzeldent, 2024-09-08:
Hi eagl,

Looks like a nice display! I assume https://www.aliexpress.com/item/1005006484114641.html.

It is preinstalled with micro-python, an esp32-s3, tn panel, however at first glance I could not find more specifications/schematics/graphics driver. Adding this board should then (if there is no new display driver required) limited to adding a definition file.

You could just take the ```ESP32-8048S043C.json``` and copy this to ```ESP32-EA7013.json``` and do some testing in there. Remember to rebuild the library (remove in .pio directory) after making changes to the definitions.

Goog luck and I can add the definition if it is working!

Can you attach the specifications to make a better assessment?

To integrate

eagl1, 2024-09-09:
To me I got it for half the price in the Aliexpress link you provided, and shipped to me to Saudi Arabia.

Yes, I got it to work first in the ESP-IDF, then I copied the pin assignment to the Arduino code and it worked.

One time the module didn't work and also didn't have boot button, so I had to pull IO0 to GND manually and it worked and I was able to reprogram it.

[EA7013.pdf](https://github.com/user-attachments/files/16937523/EA7013.pdf)
[亦亞徽EA7013-S3-TN规格书.pdf](https://github.com/user-attachments/files/16937524/EA7013-S3-TN.pdf)

The attached files are all I got from the seller.
I think it's very similar to the yellow 7" on Aliexpress.

eagl1, 2024-09-14:
OK, I got the necessary settings to get it to work on Arduino. Based on `GFX_Library_for_Arduino-1.4.x`:

1. Install  ESP32 board manager from espriff 2.0.17. If you want to work with `ESP32_Display_Panel-master` then ESP32 3.0.X board package should work.
2. Enable PSRAM OPI.
3. CH340 may not be important but CH340 2019 version worked with me.
4. Maybe you might need to press boot button maybe at least once, in my case I must pull IO0 to GND manaully.

This also will work with `ESP32_8048S043` display too.

eagl1, 2024-09-14:
These the pin settings from ESP-IDF:

```
///////////////////////////////////////////////////////////////////////////////////////////////////////////////////
// ESP-IDF IDE RGB setting
// 7"
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (15 * 1000 * 1000)
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_BK_LIGHT       1
#define EXAMPLE_PIN_NUM_HSYNC          4
#define EXAMPLE_PIN_NUM_VSYNC          2
#define EXAMPLE_PIN_NUM_DE             5
#define EXAMPLE_PIN_NUM_PCLK           6
#define EXAMPLE_PIN_NUM_DATA0          14 // B0
#define EXAMPLE_PIN_NUM_DATA1          21 // B1
#define EXAMPLE_PIN_NUM_DATA2          47 // B2
#define EXAMPLE_PIN_NUM_DATA3          48 // B3
#define EXAMPLE_PIN_NUM_DATA4          45 // B4
#define EXAMPLE_PIN_NUM_DATA5          46 // G0
#define EXAMPLE_PIN_NUM_DATA6          9  // G1
#define EXAMPLE_PIN_NUM_DATA7          10 // G2
#define EXAMPLE_PIN_NUM_DATA8          11 // G3
#define EXAMPLE_PIN_NUM_DATA9          12 // G4
#define EXAMPLE_PIN_NUM_DATA10         13 // G5
#define EXAMPLE_PIN_NUM_DATA11         7  // R0
#define EXAMPLE_PIN_NUM_DATA12         15 // R1
#define EXAMPLE_PIN_NUM_DATA13         16 // R2
#define EXAMPLE_PIN_NUM_DATA14         8  // R3
#define EXAMPLE_PIN_NUM_DATA15         3  // R4
#define EXAMPLE_PIN_NUM_DISP_EN        -1
```

rzeldent, 2024-09-15:
Hi Eagl1,

I made a definition for the board: https://github.com/rzeldent/platformio-espressif32-sunton/blob/ea7013/esp32-EA7013S3TN.json

If you put this board definition (or do a submodule on the ea7013 branch from https://github.com/rzeldent/platformio-espressif32-sunton) you can use it.

The settings for the display are in there but do not known the display timings. e.g. TIMINGS_HSYNC_PULSE_WIDTH etc... So do not expect it to work. However, you can modify the values there and recompile and tinker a bit.
I added the speaker but the touch and TF also need to be defined.
As I do not own the board it is hard to help but at least you have a starting point!

Good luck and let me know about your progress!

eagl1, 2024-09-15:
I wish I posted the Arduino ESP32RGGBPanel object, it has the other necessary settings:

```
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    5 /* DE */, 2 /* VSYNC */, 4 /* HSYNC */, 6 /* PCLK */,
    7 /* R0 */, 15 /* R1 */, 16 /* R2 */, 8 /* R3 */, 3 /* R4 */,
    46 /* G0 */, 9 /* G1 */, 10 /* G2 */, 11 /* G3 */, 12 /* G4 */, 13 /* G5 */,
    14 /* B0 */, 21 /* B1 */, 47 /* B2 */, 48 /* B3 */, 45 /* B4 */,    

    0 /* hsync_polarity */, 10 /* hsync_front_porch */, 16 /* hsync_pulse_width */, 32 /* hsync_back_porch */,
    0 /* vsync_polarity */, 15 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 12 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 15000000 /* prefer_speed */);

```

Anyway I''m working on it.

I compiled it and this is the result:
```
Environment         Status    Duration
------------------  --------  ------------
esp32-1732S019C     SUCCESS   00:00:12.240
esp32-1732S019N     FAILED    00:00:37.150
esp32-2424S012C     FAILED    00:00:21.772
esp32-2424S012N     FAILED    00:00:21.425
esp32-2432S022N     FAILED    00:00:21.160
esp32-2432S022C     FAILED    00:00:20.015
esp32-2432S024C     FAILED    00:00:19.229
esp32-2432S024N     FAILED    00:00:18.844
esp32-2432S024R     FAILED    00:00:20.040
esp32-2432S028R     FAILED    00:00:20.489
esp32-2432S028Rv2   FAILED    00:00:19.502
esp32-2432S028Rv3   FAILED    00:00:18.780
esp32-2432S032C     FAILED    00:00:19.833
esp32-2432S032N     FAILED    00:00:24.266
esp32-2432S032R     FAILED    00:00:21.680
esp32-3248S035C     SUCCESS   00:00:18.613
esp32-3248S035R     FAILED    00:00:19.805
esp32-4827S043C     FAILED    00:00:18.718
esp32-4827S043R     FAILED    00:00:18.004
esp32-4827S043N     FAILED    00:00:19.959
esp32-4848S040CIY1  FAILED    00:00:19.234
esp32-4848S040CIY3  FAILED    00:00:19.103
esp32-8048S043C     SUCCESS   00:00:08.238
esp32-8048S043N     FAILED    00:00:18.510
esp32-8048S043R     FAILED    00:00:17.936
esp32-8048S050C     FAILED    00:00:18.998
esp32-8048S050N     FAILED    00:00:17.404
esp32-8048S050R     FAILED    00:00:20.766
esp32-8048S070C     FAILED    00:00:24.045
esp32-8048S070N     FAILED    00:00:19.107
esp32-8048S070R     FAILED    00:00:19.068
esp32-EA7013        FAILED    00:00:09.104
=========================== 29 failed, 3 succeeded in 00:10:23.037 ===========================
 *  The terminal process "C:\Users\eagle\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

The ones succeeded are the ones I have except the first one and esp32-EA7013.

I think I will delete it and download a new one.

eagl1, 2024-09-15:
How to only compile specific boards ?

eagl1, 2024-09-15:
Which one I should download ?

[image]

rzeldent, 2024-09-15:
> How to only compile specific boards ?

In comment the board at the start of the file platformio.ini

rzeldent, 2024-09-15:
> Which one I should download ?
> 
> 
> 
> [image]
> 
> 

The demo application but change the sub module reference to the branch ea7013

eagl1, 2024-09-15:
> > How to only compile specific boards ?
> 
> In comment the board at the start of the file platformio.ini

You mean I comment the board I don't want to uncomment the board I want ?

eagl1, 2024-09-20:
Is the `esp32-1732S019C`, required as a dependency ?

Because when I launch VS code I get this message:

```
Resolving esp32-1732S019C dependencies...
UnknownBoard: Unknown board ID 'esp32-1732S019C'
```

[image]

As it clear I'm commenting all the boards except `esp32-8048S043C`. Am I on the right path ?

I still don't know how to compile the project at this point ?

eagl1, 2024-09-20:
Also, when I hit the correct mark to build the project I get this java installation requirement message:

[image]

How to solve this problem ? I tried to install this java but I think I didn't succeed.

rzeldent, 2024-09-26:
The # means it is already commented. Do you have the boards (json files) in the boards directory? Did you clone with --recurse submodules?

eagl1, 2024-09-28:
I downloaded both projects: `esp32-smartdisplay-main` and `esp32-smartdisplay-demo-main`. I used to have all the files and I remember the project compiled with me but now it's empty and there's no `json` files.

And when I try to compile the project I get this message:
```
Resolving esp32-8048S043C dependencies...
UnknownBoard: Unknown board ID 'esp32-8048S043C'
```

eagl1, 2024-09-29:
I'm sorry I forgot to run git command, anyway I installed the project again and added the display and copied the `json` file and pasted the settings for this display according to my editing but it didn't work. But it works on ESP IDF and arduino with no issues.

I get this output:

[image]

This is the json file:

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ARDUINO_ESP32S3_DEV'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=0'",
      "'-D ESP32_EA7013'",
      "'-D DISPLAY_WIDTH=800'",
      "'-D DISPLAY_HEIGHT=480'",
      "'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=1'",
      "'-D DISPLAY_ST7262_PAR'",
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(15*1000000)'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_H_RES=DISPLAY_WIDTH'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_V_RES=DISPLAY_HEIGHT'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_PULSE_WIDTH=16'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_BACK_PORCH=32'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_FRONT_PORCH=10'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_PULSE_WIDTH=3'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_BACK_PORCH=12'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_FRONT_PORCH=15'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_HSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_VSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_DE_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_ACTIVE_NEG=true'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_DATA_WIDTH=16'",
      "'-D ST7262_PANEL_CONFIG_SRAM_TRANS_ALIGN=4'",
      "'-D ST7262_PANEL_CONFIG_PSRAM_TRANS_ALIGN=64'",
      "'-D ST7262_PANEL_CONFIG_HSYNC_GPIO_NUM=4'",
      "'-D ST7262_PANEL_CONFIG_VSYNC_GPIO_NUM=2'",
      "'-D ST7262_PANEL_CONFIG_DE_GPIO_NUM=5'",
      "'-D ST7262_PANEL_CONFIG_PCLK_GPIO_NUM=6'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=15'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=16'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=8'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=3'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=46'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=9'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=10'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=11'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=12'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=13'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=14'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=21'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=45'",
      "'-D ST7262_PANEL_CONFIG_DISP_GPIO_NUM=GPIO_NUM_NC'",
[... 52 lines cut ...]
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-EA7013",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://vi.aliexpress.com/item/1005006484114641.html?spm=a2g0o.productlist.main.1.68452a7ddoHNgR&algo_pvid=ed5205e8-6a00-42cd-bc66-1815296e7021&algo_exp_id=ed5205e8-6a00-42cd-bc66-1815296e7021-0&pdp_npi=4%40dis%21SAR%21321.51%21225.05%21%21%21609.16%21426.41%21%402101e83717197688465127814e9977%2112000037380385588%21sea%21SA%21175575373%21&curPageLogUid=w7e1iiDOEFeX&utparam-url=scene%3Asearch%7Cquery_from%3A",
  "vendor": "https://www.eya-display.com"
}
```

I don't know where is the problem ? I tried to program it with another platformio project but got the same result, of course because I'm using the same json file, so there's must something wrong in the settings.