rzeldent/esp32-smartdisplay-demo #40: Compile Error with board = esp32-2432S032C

ForrestFire0, 2024-08-18:
Cloned project minutes ago. Using board = esp32-2432S032C I get the following issue:

```
C:\Users\forre\.platformio\penv\Scripts\pio run -t upload -e esp32-2432S032C
Processing esp32-2432S032C (board: esp32-2432S032C; platform: espressif32; frame
work: arduino)
--------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-2432S032C.html
PLATFORM: Espressif 32 (2024.6.10+sha.0b170b8) > esp32-2432S032C
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-t
iny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.3+sha.fd7a296
 - tool-esptoolpy @ 4.7.5
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.10+sha.6851dce
|-- lvgl @ 8.4.0
Building in release mode
Compiling .pio\build\esp32-2432S032C\libd1b\lvgl\draw\lv_draw_mask.c.o
Compiling .pio\build\esp32-2432S032C\libd1b\lvgl\draw\lv_img_decoder.c.o
Compiling .pio\build\esp32-2432S032C\libd1b\lvgl\draw\nxp\pxp\lv_gpu_nxp_pxp_osa.c.o

... much other stuff here ...

Compiling .pio\build\esp32-2432S032C\libf5a\esp32_smartdisplay\lvgl_touch_cst816s_i2c.c.o
Compiling .pio\build\esp32-2432S032C\libf5a\esp32_smartdisplay\lvgl_touch_gt911_i2c.c.o
Compiling .pio\build\esp32-2432S032C\libf5a\esp32_smartdisplay\lvgl_touch_xpt2046_spi.c.o
.pio/libdeps/esp32-2432S032C/esp32_smartdisplay/src/lvgl_panel_st7789_spi.c: In function 'lvgl_lcd_init':
.pio/libdeps/esp32-2432S032C/esp32_smartdisplay/src/lvgl_panel_st7789_spi.c:62:14: error: 'struct <anonymous>' has no member named 'dc_as_cmd_phase'
   62 |             .dc_as_cmd_phase = ST7789_SPI_CONFIG_FLAGS_DC_AS_CMD_PHASE,
      |              ^~~~~~~~~~~~~~~
Compiling .pio\build\esp32-2432S032C\FrameworkArduino\Esp.cpp.o
*** [.pio\build\esp32-2432S032C\libf5a\esp32_smartdisplay\lvgl_panel_st7789_spi.c.o] Error 1
============================================================================ [FAILED] Took 15.92 seconds ============================================================================

```
[image]
CLION detects that as an issue. Commenting out the offending line makes the error go away and it compiles successfully.

I have no idea how this effects the code base. I am still working out uploading so can't tell you yet if this effects functionality.

Thank you!

ForrestFire0, 2024-08-18:
can confirm the demo works when this line is commented out

rzeldent, 2024-09-08:
Did you updated the git submodules and are they pointing to the latest version?

closed 2024-11-16 by rzeldent