rzeldent/esp32-smartdisplay-demo #43: Demo tested with esp32-3248S035C ?

noeldodd, 2024-09-02:
Does the esp32-3248S035C, with it's higher resolution, have sufficient memory to run the demo? Mine hangs at `lv_display_set_buffers`

I'm not sure if this is a esp32-smartdisplay-demo, an lvgl thing, or if the board can't be used this way. When shipped it had the lvgl demo on it, and the display & cap touch worked fine so I don't think it's bum hardware.

But this demo hangs at `lv_display_set_buffers`; I added some extra log statements in lvgl to localize it:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[  5024][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  5034][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[  5045][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[  5059][I][main.cpp:27] setup(): Board: esp32-3248S035C
[  5064][I][main.cpp:28] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[  5072][I][main.cpp:29] setup(): Free heap: 245552 bytes
[  5078][I][main.cpp:30] setup(): Free PSRAM: 0 bytes
[  5082][I][main.cpp:31] setup(): SDK version: v4.4.7-dirty
[  5088][D][esp32_smartdisplay.c:159] smartdisplay_init(): smartdisplay_init
[  5095][D][esp32_smartdisplay.c:110] smartdisplay_led_set_rgb(): R:0, G:0, B:0
[  5103][I][esp32_smartdisplay.c:37] lvgl_log(): [Info] (0.000, +0)      lv_init: begin lv_init.c:160

[  5121][V][lvgl_panel_st7796_spi.c:34] lvgl_lcd_init(): display:0x3ffc2bfc
[  5128][V][lvgl_panel_st7796_spi.c:37] lvgl_lcd_init(): heap_caps_malloc w/ 3 x 38400 = 115200
[  5137][V][lvgl_panel_st7796_spi.c:40] lvgl_lcd_init(): lv_display_set_buffers...
```

I also tried changing the `LV_MEM_SIZE` calculation in `lv_conf.h`; same result:

```
#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (96 * 1024U)          /*[bytes] - was 64x, 128==error */
```

If it is supposed to work I can chase it down with lvgl.

rzeldent, 2024-09-04:
The 3248S035 have no psram so this is probably tricky to get this running.
In LVGL9 the memory allocation has been changed so might be causing runtime problems.

noeldodd, 2024-09-07:
I can confirm the unit works with the vanilla lvgl package and this config;

```cpp
#define TFT_HOR_RES   320
#define TFT_VER_RES   480
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
uint32_t draw_buf[DRAW_BUF_SIZE / 4]; 
```

noeldodd, 2024-09-07:
Also identical are the pin defs in tft_espi's `User_Setup.h` and the board def in `esp32-3248S035C.json` (except spi freq of 24000000 vs 27000000 ). It also defines the display in portrait mode, so that wasn't it.

I recreated an almost empty app but I'm still seeing a hang at `smartdisplay_init()`, which never returns. I have `#define LV_USE_LOG 1` in `lvgl_conf.h` and set to `LV_LOG_LEVEL_TRACE` but log_v et al won't log anything.

noeldodd, 2024-09-07:
May have cracked it; if I divide the buffer size by 4 in `lvgl_panel_st7796_spi.c`, (line 36-ish)

```
uint32_t drawBufferSize = sizeof(lv_color_t) * LVGL_BUFFER_PIXELS / 4;
```
I get a white display (I'm not drawing anything yet), and I can get 500ms debug prints from loop(), so it is no longer hanging. I guessed that 115200 bytes of draw buffer was too large, seeing as the draw buffer in `lv_conf`, line 108, was only 24k.

rzeldent, 2024-09-08:
Hi Noel,

I was looking at this issue and saw that there was already some reduction in the buffersize present:

In the file ```esp32-3248S035C.json``` there is already a reduction in the buffer size; only a quarter if the screen is used:
```
"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/4)'",
```

This is probably too optimistic because of the limited memory.

I propose to update the value in the ```esp32-3248S035C.json``` to something more workable. Some tweaking is required here.
So could you check out if if you change the line above (and have the original esp32-smartdisplay) to:

```
"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/16)'",
```

It also works? To test this you will have to delete the precompiled files in the .pio directory and rebuild the libraries...

noeldodd, 2024-09-09:
Ok, I did this:

- returned the line I changed in `lvgl_panel_st7796_spi.c` to NOT divide by 4
- changed the line in `esp32-3248S035C.json` to divide by 16 (from your comment above)
- deleted the build dir, recompiled, uploaded ...

...and it works just fine. Clicking on the 'rotate' button the redrawing is visible and smooth.

For giggles I changed it again to /32 (!) and it *STILL* works. It might be the smallest amount slower to redraw, but it's hard to tell without maybe recording it with a camera and counting video frames.

At /64 it still works and redrawing is still smooth - but it is slower... 1 second per screen?. So it must be efficient at shuffling those screen buffers around. Perhaps it might be more noticeable with a large-ish bitmap to redraw? I don't know how to turn on the FPS counter (yet).

In any event I think we can safely say it's tested with this hardware, which was my original question... pretty amazing, actually!

BTW, thanks for the effort on this lib; it's nice to have it consolidated like this!

If there are more tests you'd like, let me know. I've just started tinkering with squareline, so it'll be a while before the hardware shuffles off to it's new home. It's a hobby project, so no hard deadlines.

pilamrocky, 2024-09-24:
First off I have been battling the board I bought ( ESP32-3248S035C ) for weeks to get it running. Although most of that time was spent trying to get lv_micropython on it. Finding this repo has been a God send.

I also had a lot of issue trying to get past the screen loading. I started over from scratch and only made the change you suggested by changing the board setting to `"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/16)'"` like you instructed. The demo loaded fine.

pilamrocky, 2024-09-26:
While we are talking about this file I also noticed the SRAM is set to 320KB. According the AliExpress page you have linked in this file the SRAM is 520KB.

~~"maximum_ram_size": 327,680~~
"maximum_ram_size": 532480

I made this change and so far it is working.

rzeldent, 2024-09-26:
I look at platformio for the boards definitions. I think they're a bit more reliable than AliExpress...

https://registry.platformio.org/platforms/platformio/espressif32/boards

pilamrocky, 2024-09-26:
Understood. I found a few other sites that also mentioned 520KB, so I reach out to the seller for the product sheet for that specific board (ESP32-3248S035R/C) and this is what it says in the description.

> This LCD module uses ESP32-WROOM-32 module as the main control, the main
control is a dual-core MCU, integrated WI-FI and Bluetooth functions, the main
frequency can reach 240MHz, 520KB SRAM, 448KB ROM, Flash size is 4MB.

I also found a post that says maybe 520KB is the total but some of it is for IRAM?
>There is 520 KB of available SRAM (320 KB of DRAM and 200 KB of IRAM) on the ESP32.

Anyways I agree that it doesn't make sense that the board would be different then what the chip specifies that it has on it. I guess I will leave it as 320KB for now, I would just hate to miss out on memory that might be available. I don't know how to probe the board to know for sure.

Thanks for responding.

lostPixels, 2024-10-13:
I had the same issue and can confirm that adjusting this line fixed it: `"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/16)'"`

Jantleing, 2024-10-22:
> I look at platformio for the boards definitions. I think they're a bit more reliable than AliExpress...
> 
> https://registry.platformio.org/platforms/platformio/espressif32/boards

There are indeed a few boards on that site listed with 520KB because original ESP32 has 520KB of internal DRAM. First 320KB are usable as Data-/Instruction-RAM, while last 200KB are only usable as DATA-RAM.. however I am not sure if this does matter for board definitions as that instruction-caching should be handled internally by ESP?
https://en.wikipedia.org/wiki/ESP32

My esp32-3248S035C-board has a ESP32-WROOM-32-module which officially has 520KB
https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf

The Chip inside this module is identified as 
[  5049][I][main.cpp:28] setup(): CPU: **ESP32-D0WD-V3 rev3**, CPU Freq: 240 Mhz, 2 core(s)
datasheet for this Chip confirmes is has 520KB of DRAM available.
https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

I enabled full 520KB within board-definition and had no issues so far.

It might be fool-proof to leave 320KB at board definition in case vendors are forced to use other ESP32 variants, but worth to have a look at actual chip to prevent memory-bottleneck.

rzeldent, 2024-11-16:
Still not working? Can you test with the latest version?

jwitczak180, 2024-12-28:
Tested with current/latest version.  Verified the line above is as recommended (/16).  Seems like this might be a new issue (or just with my hardware) - but thought I'd chime in here first.

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[  5027][I][main.cpp:27] setup(): Board: esp32-3248S035C
[  5028][I][main.cpp:28] setup(): CPU: ESP32-D0WDQ5 rev3, CPU Freq: 240 Mhz, 2 core(s)
[  5029][I][main.cpp:29] setup(): Free heap: 276700 bytes
[  5034][I][main.cpp:30] setup(): Free PSRAM: 0 bytes
[  5039][I][main.cpp:31] setup(): SDK version: v4.4.4
[  5256][I][esp32-hal-adc.c:235] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1107

E (5262) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (5263) lcd_panel.io.spi: panel_io_spi_tx_color(380): spi transmit (queue) color failed
[  5282][E][esp_panel_st7796.c:174] st7796_draw_bitmap(): Sending RAMWR failed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40088b20
file: ".pio/libdeps/esp32-3248S035C/esp32_smartdisplay/src/lvgl_panel_st7796_spi.c" line 29
func: st7796_lv_flush
expression: esp_lcd_panel_draw_bitmap(panel_handle, area->x1, area->y1, area->x2 + 1, area->y2 + 1, px_map)

abort() was called at PC 0x40088b23 on core 1`