rzeldent/esp32-smartdisplay-demo #45: Compile error on Platformio board esp32-8048S070C

Ju1ien, 2024-10-10:
Hi,
I'm trying to compile with no success. I have cloned the repo and enabled the right borad in the .ini file

default_envs = esp32-8048S070C

When I compile, i always get the same error: 
`.pio/libdeps/esp32-8048S070C/lvgl/src/drivers/display/tft_espi/../../../display/../draw/../misc/../font/../misc/cache/../../osal/lv_os.h:20:10: fatal error: ../lv_conf_internal.h: No such file or directory
 #include "../lv_conf_internal.h"`

The file is located in the folder esp32-smartdisplay-demo\.pio\libdeps\esp32-8048S070C\lvgl\src
If I open the file lv_os.h, there is no error and the link to the #include "../lv_conf_internal.h" is correct.

Any idea what is going wrong?
Thanks!
Julien

Ju1ien, 2024-10-10:
Well, issue is fixed by moving the project folder to a shorter and no-space path, exemple:
C:\pio\esp32-smartdisplay-demo
Issue is more linked to PIO than your code. Thanks again for sharing this, very usefull starting point!
I plan on coding with EEZ studio instead of Squareline, do you have any working example or is the same principle (copy ui folder)?

rkalwak, 2024-10-30:
HI @rzeldent 

In advance big thanks for the job you put into this project! 

I am also trying to compile demo project for this esp32-8048S070C board but no success. I get the following errors when using https://github.com/rzeldent/esp32-smartdisplay#main lib:

`Compiling .pio\build\esp32-8048S070C\lib158\esp32_smartdisplay\lvgl_panel_st7262_par.c.o
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/esp32_smartdisplay.c: In function 'smartdisplay_init':
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/esp32_smartdisplay.c:197:15: error: 'lv_display_t' has no member named 'sw_rotate'
  197 |   if (!display->sw_rotate)
      |               ^~
Compiling .pio\build\esp32-8048S070C\lib158\esp32_smartdisplay\lvgl_panel_st7701_par.c.o
*** [.pio\build\esp32-8048S070C\lib158\esp32_smartdisplay\esp32_smartdisplay.c.o] Error 1
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c: In function 'lvgl_lcd_init':
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:72:12: error: 'lv_display_t' has no member named 'sw_rotate'
   72 |     display->sw_rotate = 1;
      |            ^~
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:104:10: error: 'esp_lcd_rgb_panel_config_t' has no member named 'on_frame_trans_done'      
  104 |         .on_frame_trans_done = direct_io_frame_trans_done,
      |          ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:104:32: warning: initialization of 'int' from '_Bool (*)(struct esp_lcd_panel_t *, esp_lcd_rgb_panel_event_data_t *, void *)' makes integer from pointer without a cast [-Wint-conversion]
  104 |         .on_frame_trans_done = direct_io_frame_trans_done,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:104:32: note: (near initialization for 'rgb_panel_config.data_gpio_nums[0]')
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:105:10: error: 'esp_lcd_rgb_panel_config_t' has no member named 'user_ctx'
  105 |         .user_ctx = display,
      |          ^~~~~~~~
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:105:21: warning: initialization of 'unsigned char:1' from 'lv_display_t *' makes integer from pointer without a cast [-Wint-conversion]
  105 |         .user_ctx = display,
      |                     ^~~~~~~
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:105:21: note: (near initialization for 'rgb_panel_config.flags.disp_active_low')
.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:106:82: error: 'struct <anonymous>' has no member named 'relax_on_idle'
  106 |         .flags = {.disp_active_low = ST7262_PANEL_CONFIG_FLAGS_DISP_ACTIVE_LOW, .relax_on_idle = ST7262_PANEL_CONFIG_FLAGS_RELAX_ON_IDLE, .fb_in_psram = ST7262_PANEL_CONFIG_FLAGS_FB_IN_PSRAM}};`

If I run for the develop version as it is in the repository the first error is gone but the rest is present. Would you mind to try to compile it yourself?

rzeldent, 2024-10-30:
Take a look at https://github.com/rzeldent/esp32-smartdisplay/issues/208

LVGL has modified their structures, so that's why it is failing

closed 2024-11-16 by rzeldent