rzeldent/esp32-smartdisplay-demo #47: 'lv_display_t' {aka 'struct lv_display_t'} has no member named 'sw_rotate'

klayus, 2024-10-28:
i cloned the example, set the enviroment to esp32-2432S028Rv3, selected my board - the same and this is the error i'm getting.
'lv_display_t' {aka 'struct lv_display_t'} has no member named 'sw_rotate'
if i comment out those lines in esp32_smartdisplay.c - if (!display->sw_rotate)
and in lvgl_panel_st7789_spi.c  display->sw_rotate = 0;

the example works but i cannot rotate the example. the image becomes garbled :)

bobskigit, 2024-10-28:
As of 18th September, sw_rotate was removed from lvgl. Unfortunately, no guidance is given about what to replace it with

https://github.com/lvgl/lvgl/commit/1dd63c7e5fb48399c77109da9edcd57c0f7c5abc

rzeldent, 2024-10-28:
For now I pinned it to LVGL 9.2.0: See Demo broken again ·
rzeldent/esp32-smartdisplay · Discussion #209
<https://github.com/rzeldent/esp32-smartdisplay/discussions/209>

Unclear what LVGL is doing? Is it removed?

bobskigit, 2024-10-28:
Yes , it has been removed. The commit message says

**fix(display): remove the unused sw_rotate field**

klayus, 2024-10-29:
It worked using:
lib_deps =
   https://github.com/rzeldent/esp32-smartdisplay#develop
   lvgl/lvgl@^9.2.0

Thank you

closed 2024-10-29 by klayus