rzeldent/esp32-smartdisplay-demo #48: esp32-2432W328C rotation button not working as expected

lcscosta, 2024-10-31:
Hello guys,

I have an ESP32-2432W328C, using this demo, the button Add, CDR and built-in Led are working as expected.

But the rotation button creates a bug. I believe that changes in sw_rotate(LVGL 9.2.2) might be causing this resolution problem with the rotate button. I've tried some solutions but none worked. If someone can share your experience using this board in Landscape mode, I will appreciate it.

The following pictures showing the bug:

[image]

[image]

rzeldent, 2024-11-16:
Can you test this is resolved in the latest release? [2.1.0](https://github.com/rzeldent/esp32-smartdisplay/releases/tag/2.1.0)

lcscosta, 2024-11-19:
Problem solved, thank you very much!!!

[image]

closed 2024-11-19 by lcscosta