rzeldent/esp32-smartdisplay-demo #54: Demo doesn't work on v9.2.2 and JC8048W550 board

bobcroft, 2024-12-29:
The project fails to compile on the latest version of VSC / platformIO with the following persistent error;
[image]

The version of LVGL is 9.2.2.  When I tried to use v9.1.0 it raised an error with board file incompatibility.  It seems  there are errors / issues with the lvgl_panel_st7262.par.c file.  Any idea how to fix this please.

rzeldent, 2024-12-29:
Hi bobcroft,

The versions  of esp-smartdisplay are tied to a version of LVGL. So use the v9.2.2 version. I also suggest to use the develop branch where new boards that I do not own are under development...

bobcroft, 2024-12-29:
Hi Rene,

I am using v9.2.2 as that is the version that is loaded when I instal land run the demo code.  I also tried another project that found, using your library, and I get exactly the same error, using 9.2.2.

How would I use the development branch please?  I am not very familiar with Github, sorry.

Bob

Jeronimo59, 2025-07-09:
Hello @rzeldent, and many thanks for the good work.

I am facing the same compiler issues as @bobcroft  when trying your demo with ESP32-8048S050, using 9.2.2

`CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-8048S050C.html
PLATFORM: Espressif 32 (54.3.20) > esp32-8048S050C
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - tool-esptoolpy @ 4.8.9 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 14.2.0+20241119 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 9.2.2
|-- esp32_smartdisplay @ 2.1.1+sha.327322c
Building in release mode
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\esp32_smartdisplay.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_axa15231b_qspi.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_gc9a01_spi.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_ili9341_spi.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_st7262_par.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_st7701_par.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_st7789_i80.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_st7789_spi.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_panel_st7796_spi.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_touch_cst816s_i2c.c.o
In file included from .pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c:1:
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/include/esp32_smartdisplay.h:44:91: error: unknown type name 'uint'; did you mean 'int'?
   44 |     void smartdisplay_lcd_set_brightness_cb(smartdisplay_lcd_adaptive_brightness_cb_t cb, uint interval);
      |                                                                                           ^~~~
      |                                                                                           int
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/esp32_smartdisplay.c:92:87: error: unknown type name 'uint'; did you mean 'int'?
   92 | void smartdisplay_lcd_set_brightness_cb(smartdisplay_lcd_adaptive_brightness_cb_t cb, uint interval)
      |                                                                                       ^~~~ 
      |                                                                                       int  
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_touch_gt911_i2c.c.o
Compiling .pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\lvgl_touch_xpt2046_spi.c.o
*** [.pio\build\esp32-8048S050C\lib002\esp32_smartdisplay\esp32_smartdisplay.c.o] Error 1
In file included from .pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:3:
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/include/esp32_smartdisplay.h:44:91: error: unknown type name 'uint'; did you mean 'int'?
   44 |     void smartdisplay_lcd_set_brightness_cb(smartdisplay_lcd_adaptive_brightness_cb_t cb, uint interval);
      |                                                                                           ^~~~
      |                                                                                           int
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c: In function 'lvgl_lcd_init':
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:101:10: error: 'esp_lcd_rgb_panel_config_t' has no member named 'on_frame_trans_done'
  101 |         .on_frame_trans_done = direct_io_frame_trans_done,
      |          ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-8048S050C/esp32_smartdisplay/src/lvgl_panel_st7262_par.c:101:32: error: initialization of 'int' from '_Bool (*)(struct esp_lcd_panel_t *, esp_lcd_rgb_panel_event_data_t *, void *)' makes integer from pointer without a cast [-Wint-conversion]`