rzeldent/esp32-smartdisplay-demo #56: Build fails for esp32-4848S040 (and probably others)

skiphansen, 2025-02-16:
I just checkout out the demo and tried to build both esp32-4848S040CIY1 and esp32-4848S040CIY3
but both failed with

```
.pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/include/esp32_smartdisplay.h:44:91: error: unknown type name 'uint'; did you mean 'int'?
```
Full [build.log](https://github.com/user-attachments/files/18815335/build.log)

skiphansen, 2025-02-16:
This issue was created when I was using PIO
```
 skip@Dell-5510:~/esl/OpenEPaperLink/ESP32_AP-Flasher$ pio --version
PlatformIO Core, version 6.1.18a1
```

Another project which worked before also failed to build (for other reasons) and I got frustrated so
I reinstalled platformIO from scratch.  

```
skip@Dell-5510:~/open_src/esp32-smartdisplay-demo$ pio --version
PlatformIO Core, version 6.1.17
skip@Dell-5510:~/open_src/esp32-smartdisplay-demo$
```

That fixed the problem for that project so I tired this one again and the demo build successfully.  

Weird, I have NO idea !

closed 2025-02-16 by skiphansen

b-oern, 2025-04-10:
Same Issue on esp32-2432S022C