rzeldent/esp32-smartdisplay-demo #58: lvgl spinner crashes ESP32-2423S022N

jdgordon, 2025-04-16:
Hi,

This is very likely not the correct place to put this bug but hopefully you can help (or point me in the right direction).

I've taken this demo and started working with it, I wanted to add the lvgl spinner widget as a loading screen. I replaced the qrcode in the demo main.cpp with the following code and the system either locks up or crashes after about 7 seconds very reliably.

```
    ui_init();

    auto ui_Spinner1 = lv_spinner_create(ui_scrMain);
    lv_spinner_set_anim_params(ui_Spinner1, 1000, 90);
    lv_obj_set_width(ui_Spinner1, 142);
    lv_obj_set_height(ui_Spinner1, 136);
    lv_obj_set_align(ui_Spinner1, LV_ALIGN_CENTER);
    lv_obj_remove_flag(ui_Spinner1, LV_OBJ_FLAG_CLICKABLE);      /// Flags
    lv_obj_center(ui_Spinner1);

```

I'd be happy to debug with gdb but I cant figure out how to connect with openocd (assuming it even provides the debugger over usb-c which i'm not sure it does).

Thanks

rzeldent, 2025-04-17:
Hi Jonathan,

Looks good. Is there maybe a define to include the spinner functionality?

jdgordon, 2025-04-17:
@rzeldent hi, no, no extra headers needed