rzeldent/esp32-smartdisplay-demo #59: ESP32-8048S043 stuck in colour loop

freedirect, 2025-04-27:
I have a problem with the ESP32-8048S043 screen, when I load the demo (as it is) all I get is a loop of colour screens that change every 3 seconds, first black, then white, red, green, blue and so on.

What I have done with visual studio is to execute in the terminal 

**"git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo"** 

and then uncomment the line:

**"default_envs = esp32-8048S043C" in the file "platformio.ino"**

It doesn't really seem that there is any error in the compilation, and the truth is that it loads on the board correctly. In fact, the serial port shows the following:

_ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[   137][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5148][I][main.cpp:27] setup(): Board: esp32-8048S043C
[  5153][I][main.cpp:28] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s) 
[  5161][I][main.cpp:29] setup(): Free heap: 300564 bytes
[  5166][I][main.cpp:30] setup(): Free PSRAM: 8386279 bytes
[  5171][I][main.cpp:31] setup(): SDK version: v4.4.7-dirty
[  5253][W][esp_touch_gt911.c:162] gt911_read_info(): Resolution obtained from GT911 (480,272) does not match resolution (800,480). Enabled coordinate adjustment._

I don't know if you can give me a hand to be able to start with this, thank you very much.

gibellinim, 2025-05-26:
i had the same issue with a JC8048W550C i used the json found below and now it works

https://github.com/rzeldent/platformio-espressif32-sunton/blob/main/JC8048W550.json

rzeldent, 2026-08-25:
Updated in JC8048W550C.json

closed 2026-08-25 by rzeldent